nodemcu/nodemcu-firmware #825: Noob question - cannot connect

ylh888, 2015-12-08:
I have a Nodemcu. When I plug the USB into my Mac, am I supposed to see the board show up on the USB port as serial?

Using ESPlorer, how do I connect to the board if I don't see it?

Or, am I supposer to see it on Wifi?  (netstat -r does not show)

If I use Arduino IDE, the board does not show up either (I've added the Nodemcu module to the Board Manager and can see the examples).

If there is already onboard Lua environment, I assume I don't need to flash the firmware. Is this true?

Please help me get through this initial hurdle.

marcelstoer, 2015-12-08:
Please see #719.

closed 2015-12-08 by marcelstoer

nickandrew, 2015-12-09:
README.md should answer these basic getting-started type questions, and if somebody tells me what the tty device name is on a Mac when it's plugged in, I'll add that. But see #719; NodeMCU issues aren't for answering these questions.

ylh888, 2015-12-09:
I apologize. I got a new “nodemcu” board but it was running the AT command set and I had no idea. Didn’t know that I had to flash the firmware first.

Your git project is the first place nodemcu.org points to and there was scant mention in nodemcu.org of how one should bootstrap oneself.

One piece of information might be useful in the README for people who run OSX El Capitan:

This piece of driver is critical (other driver may not work)  http://www.wemos.cc/wiki/doku.php?id=en:ch340g

Without it the board appears to be dead and doesn’t show up at all.

Thanks!

marcoskirsch, 2015-12-09:
This piece of driver is critical (other driver may not work)  http://www.wemos.cc/wiki/doku.php?id=en:ch340g

It sounds like you have an old NodeMCU. The 1.0 dev kit uses a USB->serial chip that works with OS X out of the box.

ylh888, 2015-12-09:
The board is based on 1.4.0 (Dec 2014). Apparently it is El Capitan that's broken.

marcoskirsch, 2015-12-09:
This is not related to a firmware bug, so carrying out this conversation here is the wrong thing to do because post notifications will reach everyone interested in firmware bugs. That's annoying. Let's take it to the proper forum: http://www.esp8266.com/

If your board has the ch340 serial chip, then indeed OS X does not come with drivers for it out of the box. You will need to install sketchy drivers that aren't signed (triple sketchy). My understanding was that all newer NodeMCU boards use a serial chip from a different vendor. OS X does come out of the box with drivers for that. I can confirm as I've had both generations of NodeMCU.

No more replies from me here :)