nodemcu/nodemcu-firmware #826: CONFUSE ABOUT ADC.READ

vudangngoc, 2015-12-09:
Sorry to post here, I aware about #719 and post to esp8266.com but got no answer after 2 days :(
I use a LM35 sensor, wired it with 5V power and I got 0.23 Vout. It's correct with my room temp( 10mV per Celsius temp). But when I wire pin out with ADC pin of esp7 and call print(adc.read(0)), console print out 167. I cannot find relationship between 0.23V and result 167 of adc.read. I found some article said that ESP's ADC have range 0 ~ 1V input, in this case adc.read should return 0.23*1024 = 235.
By the way, is there any reason cause adc.read alway return 65535?
Thank you for any support in advance

eyaleb, 2015-12-09:
Now *I* am confused. Does `adc.read(0)` give you `167` or `65535`?
The latter indicates a failure (it is unsigned `0xffff`). Could be the voltage is above 1V. Or a wrong fw.

My understanding is that with the latest dev/master you need to build the firmware with `app/include/user_config.h` saying this to enable `adc.read(0)`:
```
// Byte 107 of esp_init_data_default, only one of these 3 can be picked
//#define ESP_INIT_DATA_ENABLE_READVDD33
#define ESP_INIT_DATA_ENABLE_READADC
//#define ESP_INIT_DATA_FIXED_VDD33_VALUE 33
```

vudangngoc, 2015-12-09:
thank you for your support, I modify app/include/user_config.h and now adc.read(0) return 232 according to temp at my room

closed 2015-12-09 by vudangngoc

ishaan80, 2016-02-10:
Hi, I am having the same problem. I am currently trying to test voltage between 0-1 V but then ever I write print(adc.read(0))
It always returns 65535. 
I initially though its a board problem but now i have 2 boards and both of them give me the same result.
How were you able to solve it?