nodemcu/nodemcu-firmware #827: Drop the changelog

marcelstoer, 2015-12-09:
In https://github.com/nodemcu/nodemcu-firmware/issues/799#issuecomment-163195340 @nickandrew asked whether we should drop the changelog.

I fully second that but didn't want to rush ahead without having broad consensus among the committers.

nickandrew, 2015-12-09:
(In the wiki, that is. I already dropped it from README.md)

nodemcu, 2015-12-09:
+1, wiki too.

devsaurus, 2015-12-09:
How will we document and communicate the changes between releases?

marcelstoer, 2015-12-09:
If we follow the contributing guidelines then the list of PRs is the changelog.

ghusson2, 2015-12-09:
from [issue#799](https://github.com/nodemcu/nodemcu-firmware/issues/799) : 
If the changelog is not maintained up-to-date, it should be deleted I agree. As a user, I will look at the issues if I have an issue myself.
And what is important for an end user (in addition to the user manual/tutorial) is a roadmap with a following of what has been implemented (ie : what has been improved since the beginning, and what we are working on). A good exemple I know for that is Proxmox VE (simple and efficient : roadmap/release history) : https://pve.proxmox.com/wiki/Roadmap.

devsaurus, 2015-12-09:
> If the changelog is not maintained up-to-date, it should be deleted I agree.

Agree as well - outdated information is often worse than than the lack of it.
I was wondering what could replace this source. My impression is that we should be able to describe the added value of our deliveries (aka releases) to the end users.

> If we follow the contributing guidelines then the list of PRs is the changelog.

Ok, good point. As I'm still not very experienced with github I probably don't understand the all benefits of this concept yet.

> And what is important for an end user (in addition to the user manual/tutorial) is a roadmap with a following of what has been implemented.

It's an appealing idea to sketch the planned extensions and additions in a roadmap on feature level. But this is probably going beyond the intention of this issue.

TerryE, 2015-12-09:
> If we follow the contributing guidelines then the list of PRs is the changelog

+1.  Thinking about this, the changelog is woefully out of date and none has the interest or bandwidth to maintain it.  It would be better to include a link to the git commits and that way uses could browse the audit trail.  One slight variation to this:  when we implement Johny's suggestion of doing occasional snapshots of dev into master and Nick's suggestion to tag these using meaningful labels, it might be worth adding a paragraph summary of the main changes since the last tag; not a full list but more just a high-level summary.

marcelstoer, 2015-12-09:
> it might be worth adding a paragraph summary of the main changes since the last tag; not a full list but more just a high-level summary.

Yes, definitely. Git/GitHub does give you that. It'd be like e.g. https://github.com/nodemcu/nodemcu-firmware/releases/tag/0.9.6-dev_20150627 (without the binaries for download). Sidenote: if those tags are not associated with a release they need to be created locally and then "tag-pushed" using `git push origin --tags`.

nickandrew, 2015-12-10:
Ok, I have edited the wiki and replaced the changelog with a link to the closed PRs.

In #820 I asked for the repository to be tagged, so `git describe` no longer describes itself as 0.9.6

Per @TerryE, putting a summary of the main changes since the last tag into each annotated tag message is a good idea. If somebody wants to do that for 1.4.0 then please go ahead, otherwise you can just grab this tag https://github.com/nickandrew/nodemcu-firmware/releases/tag/1.4.0-20151201 and push it.

Re the ProxMox Roadmap mentioned by @ghusson2 I see a roadmap as an outline of planned future development, but this document is just a changelog of what they already did. This can be covered, as stated, by PRs for the detailed changes and one high level summary per tag/release. Maintaining any separate changelog will be troublesome; nobody wants to do it and results will be patchy and inconsistent, as we saw.

One suggestion I think nobody has mentioned yet is to maintain RELEASE_NOTES.md and update it once per tag/release. That file will document API changes important to developers upgrading between releases.

karrots, 2015-12-10:
Thoughts on using something to just reformat the commit log into a nice change log (filter merges and such)?

https://skywinder.github.io/github-changelog-generator/

That said a raw dump type of change log may be just too much noise. I think that there is something to be said for a change log that highlights major API changes though. It will go along way towards reducing issues being logged. Look at all the issues from people upgrading to a version that the ADC code changed or the wifi event monitoring for that matter.

If I upgrade from version A to version B what will I need to watch out for and what will I gain?

nickandrew, 2015-12-10:
> If I upgrade from version A to version B what will I need to watch out for and what will I gain?

@karrots Release Notes will cover the former, and comment on the release tag will cover the latter.

karrots, 2015-12-10:
OK good point

closed 2015-12-12 by marcelstoer