nodemcu/nodemcu-firmware #828: http PUT does not close as version 0.9.5 did

stefanbode, 2015-12-09:
The current build (dev) and (master) behave different than the last "official" version 0.9.5 regarding http:send. IF executed step by step in the console I got HTTP/1.1 200 OK but in a lua script the connection did not get closed and therefore the value does not get updated.

Here the small script, that interactively run into the "receive" option but never, if you put it into a .lua.

conn = net.createConnection(net.TCP, 0)
conn:on("receive", function(conn, payload) print("got reply\r\n") print(payload) end)
conn:connect(8080, "192.168.178.29");
conn:send("PUT /rest/items/Sensor_Temperature2/state HTTP/1.1\r\n")
conn:send("Host: 192.168.178.42\r\n") 
conn:send("Content-Type: text/plain\r\n")
conn:send("Content-Length: 4\r\n")
conn:send("\r\n")
conn:send("24.7\r\n")

I also try to wrap the send into a 
conn:on("connection", function(conn, payload)
	print("Got connection...") ......

But also this connection got not closed by the HOST. It remains open. I assume there was a fix since march that requires a different approach.

stefanbode, 2015-12-09:
Current Firmware:
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 68eadaaf4596a112144c3b2e1dc52a04c253601b
	SSL: false
	modules: node,file,gpio,wifi,net,tmr,uart,ow
 build 	built on: 2015-12-09 12:36
 powered by Lua 5.1.4 on SDK 1.4.0

nickandrew, 2015-12-09:
I think you need to put your `conn:send()` calls into a single call, and run that inside `conn:on("connection")`

TerryE, 2015-12-09:
@stefanbode, Sorry Stefan, but the issue here is that you haven'tread the documentation and aren't coding your Lua correctly.  There is no specific firmware bug here, so this falls under #719.

closed 2015-12-09 by TerryE