nodemcu/nodemcu-firmware #829: documentation directory ?

ghusson2, 2015-12-09:
As the project is rethinking the documentation, maybe it could be the right time for reorganization ?
For example a documentation directory with :
doc
- file history (project history, hardware, persons involved)
- file readme
- file howto_begin
- file howto_contribute (pull requests proposal, donate, official hardware maybe ?, CONTRIBUTING.md)
- file official_faq (from unofficial FAQ :)
- file links (with read further, blog tutorials and examples, ...)
- dir examples (merge/reorganize example and lua_examples and lua_modules ?)
- dir tutorial (maybe one day populated ?)

marcelstoer, 2015-12-09:
Will be part of #774 .

ghusson2, 2015-12-09:
Ok, i thought RTD was just in code markdown but it is a doc building system. I should move my comment to issue #774 and delete this issue #829 ?

marcelstoer, 2015-12-09:
You can leave it here. We can't fix all at once anyway and will work in small steps.

marcelstoer, 2016-01-17:
See https://github.com/nodemcu/nodemcu-firmware/tree/dev/docs and http://nodemcu.readthedocs.org/en/dev/.

closed 2016-01-17 by marcelstoer