nodemcu/nodemcu-firmware #833: Bug in HTTP?

chegongcai, 2015-12-10:
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 68eadaaf4596a112144c3b2e1dc52a04c253601b
	SSL: true
	modules: node,file,gpio,wifi,net,i2c,spi,tmr,adc,uart,ow,bit,mqtt,coap,u8g,cjson,crypto,rc,rtcmem,rtctime,rtcfifo,sntp,ucg,enduser_setup,hx711
 build 	built on: 2015-12-10 03:27
 powered by Lua 5.1.4 on SDK 1.4.0

when I do the test:
srv=net.createServer(net.TCP,500)
srv:listen(80,function(conn)
 conn:on("receive",function(conn,payload)
conn:send("HTTP/1.0 200 OK\r\nContent-type: text/html\r\nServer: ESP8266\r\n\r\n")
conn:send("IOT")
conn:send("<br></br>")
conn:send("by chegongcai")

I conect to the 192.168.4.1,through the Fiddler Web Debugger ，It seem just do the"conn:send("HTTP/1.0 200 OK\r\nContent-type: text/html\r\nServer: ESP8266\r\n\r\n")",and the follow three "conn:send" are don't work. Is it a  bug in HTTP?

chegongcai, 2015-12-10:
anyone can help?

vudangngoc, 2015-12-10:
Is that all your code?

chegongcai, 2015-12-10:
@vudangngoc No ,  it is a part of my code.

vudangngoc, 2015-12-10:
Connection will automatic close if you have this line of code:
conn:on("sent", function(conn) conn:close() end)

chegongcai, 2015-12-10:
there are all of my test code:
(1)routerset.lua
wifi.setmode(wifi.SOFTAP)
--set ap ssid and pwd
cfg={}
cfg.ssid="ESP8266"
cfg.pwd="12345678"
wifi.ap.config(cfg)
print(wifi.ap.getip())

--  http server
srv=net.createServer(net.TCP)
srv:listen(80,function(conn)
 conn:on("receive",function(conn,payload)
  --print(payload)
  --conn:send("<h1> Hello, NodeMcu.</h1>")
  setstate="..."
  k=string.find(payload,"ssid")
  if k then
   str=string.sub(payload,k)
   --print(str)
   m=string.find(str,"&")
   --print(m)
   strssid='ssid='..'"'..string.sub(str,6,(m-1))..'"'
   print(strssid)
   strpass='password="'..string.sub(str,(m+10))..'"'
   print(strpass)

   file.open("routerpass.lua","w+")
   file.writeline(strssid)
   file.writeline(strpass)
   file.close()

   setstate="OK"
   print("store ok")
   --file.open("routerpass.lua","r")
   --print(file.readline())
   --print(file.readline())
   --file.close()
  end

  -- html-output
  conn:send("HTTP/1.0 200 OK\r\nContent-type: text/html\r\nServer: ESP8266\r\n\n")
  conn:send("<html><head></head><body>")
  conn:send("Served from ESP8266-".. node.chipid().." - NODE.HEAP: ".. node.heap().." ")
  conn:send("<font color=\"red\">")
  conn:send("<h1> connect your router "..setstate.."</h1></font>")   --"..setstate.."
  conn:send("<FORM action=\"\" method=\"POST\">")
  conn:send("SSID:<input type=\"test\" name=\"ssid\">")
  conn:send("<br></br>")
  conn:send("Password<input type=\"text\" name=\"password\">")
  conn:send("<br></br>")
  conn:send("<input type=\"submit\" value=\"Submit\">")
  conn:send("</form>")
  conn:send("</body>")
  conn:send("</html>")

 end)
end)

chegongcai, 2015-12-10:
(2)routerpass.lua
ssid="ESP8266"
password="12345678"

chegongcai, 2015-12-10:
(3)int.lua
Set_Pin=2 --GPIO4 low for set the wifi connect to your router
--dont use GP0 GP2 GP16
gpio.mode(Set_Pin,gpio.INPUT)

if gpio.read(Set_Pin)==1 then -- if not set mode
require("routerpass") -- router ssid password store here

if true then  --change to if true
      -- require('hx711_test')
	print("set up wifi mode")
	wifi.setmode(wifi.STATION)
	--please config ssid and password according to settings of your wireless router.
	wifi.sta.config(ssid,password)
	wifi.sta.connect()
	cnt = 0
	tmr.alarm(1, 1000, 1, function() 
	    if (wifi.sta.getip() == nil) and (cnt < 20) then 
	    	print("IP unavaiable, Waiting...")
	    	cnt = cnt + 1 
	    else 
	    	tmr.stop(1)
	    	if (cnt < 20) then print("Config done, IP is "..wifi.sta.getip())
	    	--dofile("yourfile.lua")
	    	else print("Wifi setup time more than 20s, Please verify wifi.sta.config() function. Then re-download the file.")
	    	end
	    end 
	 end)
else
	print("\n")
	print("Please edit 'init.lua' first:")
	print("Step 1: Modify wifi.sta.config() function in line 5 according settings of your wireless router.")
	print("Step 2: Change the 'if false' statement in line 1 to 'if true'.")

end
else
require('routerset')
print("please open 192.168.4.1 then input ssid and password")
end

chegongcai, 2015-12-10:
I have used the Fiddler to capture, just the first "conn:send" work.

chegongcai, 2015-12-10:
@vudangngoc  Is it any problem here ？

vudangngoc, 2015-12-10:
Did you try to use this module?
https://github.com/nodemcu/nodemcu-firmware/blob/master/lua_modules/http/http.lua

chegongcai, 2015-12-10:
No,but I have tied my test code on dev0.9.6, it is ok !

TerryE, 2015-12-10:
See #719

closed 2015-12-10 by TerryE