nodemcu/nodemcu-firmware #836: Updating CONTRIBUTING.md for issues?

jmattsson, 2015-12-10:
Seeing as [CONTRIBUTING.md](https://github.com/nodemcu/nodemcu-firmware/blob/master/CONTRIBUTING.md) is linked to reasonably prominently whenever one opens a new issue, should we perhaps pre-emptively link to #719 as one of the first things in that doc?

Currently the contrib doc is very nicely geared towards developers who are wishing to raise PRs, but I think we have even more people raising issues.

Thoughts? @marcelstoer ?

vowstar, 2015-12-10:
I think it's a great idea. Contributors not only including developers, but also users. A user guideline is needed for report issues.

nodemcu, 2015-12-11:
I would recommend put the issue guideline from #719 in a file like CONTRIBUTING.md, and put link to these files in TOP of readme.md.

marcelstoer, 2015-12-11:
I partially agree.

Those creating issues asking for user support are not actually _contributing_ anything but that may be nothing but a technicality.

Whenever you open issue or a PR GitHub displays a box at the top of the input fields saying
> Please review the guidelines for contributing to this repository.

and linking to `CONTRIBUTING.md` (it should be more prominent but that's a different story). Hence, all information that we want to pass along in such a case should be contained in that file. I was really happy when @TerryE created that issue but I always thought it odd that the information should live in an issue rather than in `README.md` or `CONTRIBUTING.md`.
So, my proposal would be to transfer part, or all, of 719 to  `README.md` and part, or nothing, to `CONTRIBUTING.md`.

marcelstoer, 2016-04-10:
Will be "fixed" with #1077.

closed 2016-04-10 by marcelstoer