nodemcu/nodemcu-firmware #840: NodeMCU crashes during GPIO bitbanging

robertfoss, 2015-12-11:
I've been working on an module for the APA102 LED chipset, but have run into issues with NodeMCU crashing during bitbanging of the protocol over the GPIO pins. I've removed as much code as possible while still retaining the failure case.

The code has been tested on 2 NodeMCU modules and a ESP-01 module with the same results.
The modules are being run unconnected to anything but a serial interface and a power source.
Alternative GPIO pins have been tested with the same results.
#define NODE_DEBUG does not provide any extra details as to why the crash occurs.

As a not entirely relevant note, I've also tried running essentially the same code on Arduino-esp8266, and it works flawlessly there.

The code is available below as _apa102_test.c_ and _apa102_test.patch_

**Output of _apa102_test.start(1)_**

```
NodeMCU 1.4.0 build 20151006 powered by Lua 5.1.4 on SDK 1.4.0
lua: cannot open init.lua
> apa102_test.start(1)
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free_heap=42536
apa102_test_start send uint32_t free������1����H1�������

NodeMCU 1.4.0 build 20151006 powered by Lua 5.1.4 on SDK 1.4.0
lua: cannot open init.lua
```

**apa102_test.c**

```
static int ICACHE_FLASH_ATTR apa102_test_start(lua_State* L) {
  uint8_t data_pin = luaL_checkinteger(L, 1);

  platform_gpio_mode(data_pin, PLATFORM_GPIO_OUTPUT, PLATFORM_GPIO_FLOAT);

  while (true) {
    c_printf("apa102_test_start send uint32_t free_heap=%u\n", system_get_free_heap_size());
    ets_intr_lock();
    platform_gpio_write(data_pin, 0);
    ets_intr_unlock();
  }
}

#define MIN_OPT_LEVEL 2
#include "lrodefs.h"
const LUA_REG_TYPE apa102_test_map[] =
{
  { LSTRKEY( "start" ), LFUNCVAL( apa102_test_start )},
  { LNILKEY, LNILVAL}
};

LUALIB_API int luaopen_apa102_test(lua_State *L) {
  LREGISTER(L, "apa102", apa102_test_map);
  return 1;
}
```

**apa102_test.patch**

```
diff --git a/app/include/user_config.h b/app/include/user_config.h
index 65a80b9..cfb82c7 100644
--- a/app/include/user_config.h
+++ b/app/include/user_config.h
@@ -28,6 +28,7 @@
 
 
 #define NODE_ERROR
+//#define NODE_DEBUG
 
 #ifdef NODE_DEBUG
 #define NODE_DBG c_printf
diff --git a/app/include/user_modules.h b/app/include/user_modules.h
index 1a4646e..c3a6c33 100644
--- a/app/include/user_modules.h
+++ b/app/include/user_modules.h
@@ -16,10 +16,12 @@
 
 #ifdef LUA_USE_MODULES
 #define LUA_USE_MODULES_ADC
+#define LUA_USE_MODULES_APA102_TEST
 #define LUA_USE_MODULES_BIT
 //#define LUA_USE_MODULES_BMP085
-#define LUA_USE_MODULES_CJSON
-#define LUA_USE_MODULES_COAP
+//#define LUA_USE_MODULES_CJSON
+//#define LUA_USE_MODULES_COAP
 #define LUA_USE_MODULES_CRYPTO
 #define LUA_USE_MODULES_DHT
 //#define LUA_USE_MODULES_ENDUSER_SETUP // USE_DNS in dhcpserver.h needs to be enabled for this module to work.
@@ -27,20 +29,20 @@
 #define LUA_USE_MODULES_GPIO
 //#define LUA_USE_MODULES_HX711
 #define LUA_USE_MODULES_I2C
-#define LUA_USE_MODULES_MQTT
-#define LUA_USE_MODULES_NET
+//#define LUA_USE_MODULES_MQTT
+//#define LUA_USE_MODULES_NET
 #define LUA_USE_MODULES_NODE
-#define LUA_USE_MODULES_OW
-#define LUA_USE_MODULES_PWM
-#define LUA_USE_MODULES_RC
-#define LUA_USE_MODULES_RTCFIFO
-#define LUA_USE_MODULES_RTCMEM
-#define LUA_USE_MODULES_RTCTIME
-#define LUA_USE_MODULES_SNTP
+//#define LUA_USE_MODULES_OW
+//#define LUA_USE_MODULES_PWM
+//#define LUA_USE_MODULES_RC
+//#define LUA_USE_MODULES_RTCFIFO
+//#define LUA_USE_MODULES_RTCMEM
+//#define LUA_USE_MODULES_RTCTIME
+//#define LUA_USE_MODULES_SNTP
 #define LUA_USE_MODULES_SPI
 #define LUA_USE_MODULES_TMR
 #define LUA_USE_MODULES_TSL2561
-#define LUA_USE_MODULES_U8G
+//#define LUA_USE_MODULES_U8G
 #define LUA_USE_MODULES_UART
 //#define LUA_USE_MODULES_UCG
 #define LUA_USE_MODULES_WIFI

diff --git a/app/modules/modules.h b/app/modules/modules.h
index 190787c..94985f4 100644
--- a/app/modules/modules.h
+++ b/app/modules/modules.h
@@ -149,6 +149,14 @@
 #define ROM_MODULES_APA102
 #endif
 
+#if defined(LUA_USE_MODULES_APA102_TEST)
+#define MODULES_APA102_TEST      "apa102_test"
+#define ROM_MODULES_APA102_TEST  \
+    _ROM(MODULES_APA102_TEST, luaopen_apa102_test, apa102_test_map)
+#else
+#define ROM_MODULES_APA102_TEST
+#endif
+
 #if defined(LUA_USE_MODULES_WS2801)
 #define MODULES_WS2801      "ws2801"
 #define ROM_MODULES_WS2801  \
@@ -293,6 +301,7 @@
         ROM_MODULES_BMP085  \
         ROM_MODULES_TSL2561 \
         ROM_MODULES_HX711   \
+        ROM_MODULES_APA102_TEST
 
 #endif
diff --git a/app/modules/apa102_test.c b/app/modules/apa102_test.c
new file mode 100644
index 0000000..5cf374c
--- /dev/null
+++ b/app/modules/apa102_test.c
@@ -0,0 +1,36 @@
+#include "lualib.h"
+#include "lauxlib.h"
+#include "platform.h"
+#include "auxmods.h"
+#include "lrotable.h"
+#include "c_stdlib.h"
+#include "c_string.h"
+#include "user_interface.h"
+
+
+static int ICACHE_FLASH_ATTR apa102_test_start(lua_State* L) {
+  uint8_t data_pin = luaL_checkinteger(L, 1);
+
+  platform_gpio_mode(data_pin, PLATFORM_GPIO_OUTPUT, PLATFORM_GPIO_FLOAT);
+
+  while (true) {
+    c_printf("apa102_test_start send uint32_t free_heap=%u\n", system_get_free_heap_size());
+    ets_intr_lock();
+    platform_gpio_write(data_pin, 0);
+    ets_intr_unlock();
+  }
+}
+
+
+#define MIN_OPT_LEVEL 2
+#include "lrodefs.h"
+const LUA_REG_TYPE apa102_test_map[] =
+{
+  { LSTRKEY( "start" ), LFUNCVAL( apa102_test_start )},
+  { LNILKEY, LNILVAL}
+};
+
+LUALIB_API int luaopen_apa102_test(lua_State *L) {
+  LREGISTER(L, "apa102", apa102_test_map);
+  return 1;
+}
```

TerryE, 2015-12-11:
@robertfoss, Robert this really falls under #719, but I will tell you what the issue is since you are development contributor. Your `apa102_test.start()` is called from Lua and this is a callback task.  The normal guideline of _callbacks should take no longer than 10mS_ still applies.  In this case your while loop in `apa102_test.c` breaches this _and_ doesn't feed the watchdog.  The prints to the UART are synchronous and so you are dying with a WD timeout.  The garbage is just the boot diagnostics at the wrong speed.

I'll leave you to close the issue.

PS. if you include crypto, then disable SHA2 (unless you _really_ need it) in your `user_config.h`.  The rest of crypto is simply a wrapper for ROM / SDK functions. but SHA2 in implemented in your included code.  :smiley:

robertfoss, 2015-12-12:
Thanks for the quite helpful input!

closed 2015-12-12 by robertfoss

eyaleb, 2015-12-12:
@TerryE regarding your PS. I know this is rather OT here but...
I went through trying to get a small fw bin and found that enabling SSL (not SHA2) increases the size of the bin file significantly (by around 47KB) making it unsuitable for a 512KB module. Does not look like just a few wrappers?
```
#define CLIENT_SSL_ENABLE
#define GPIO_INTERRUPT_ENABLE
//#define MD2_ENABLE
//#define SHA2_ENABLE
```

jmattsson, 2015-12-12:
SSL != crypto Lua module

eyaleb, 2015-12-12:
So we are talking about the lua crypto module. Does it need SSL in the config though?

TerryE, 2015-12-12:
No.