nodemcu/nodemcu-firmware #843: NodeMCU firmware for the ESP32

TerryE, 2015-12-12:
This is based on a response that I made on nodemcu/nodemcu-devkit-v1.0#12, but really belongs here on the firmware issue list.  We know that the ESP32 I'd now stating to become available.  Some things that we do know about the esp32 (compared to the esp8266):

*    Espressif have opened a new support site: [esp32.com](http://esp32.com/), but which is accessible by registered users.
*    It runs a dual core LX108 (rather than the single core LX106 on the 8266)
*    It has a lot more RAM ?128Kb or 256Kb? rather than 48Kb
*    It has more pin-outs and a richer set of peripherals.

[image]
Very little further detail is approved by Espressif for discussion in the public domain, but the dual core / single memory space aspects mean that the simple non-OS SDK is unlikely to port well.  You need some form of centralised lock management and resource allocation as is provided by RTOS, and hence any new SDK will need to run under RTOS.   
 
There is so much that we don't know and until we do it is difficult to make concrete plans.

*    Will there be a material pricing differential between the esp32 and the esp8266?
*    What will be its availability and when?
*    Can the core developers get NDA access to the device specs?
*    Will Espressif keep the two devices in manufacture or will they be soon phasing out the esp8266?

Clearly the esp32 is the future, but to me one key issue is how long we have to maintain parallel support, because our development strategy will depend on whether this is months or more than a year.  Another is the availability of esp32 hardware.

However, one thing that I am confident about is that we should extend firmware support to the ESP32, and that we will probably need to do this over the [RTOS SDK](/espressif/ESP32_RTOS_SDK).

jmattsson, 2015-12-12:
- They've stated that the ESP32 will cost more than the ESP8266, but that the 8266 is expected to drop in price once the ESP32 is out \[citation needed\].
- ~200 beta boards are in the process of being shipped. I believe @nodemcu got theirs just yesterday.
- I'd expect so. At work we got in on the beta program as well, so once I'm back from my leave I should have some access I believe.
- My understanding is that the ESP8266 is here to stay for quite a while, as the cheaper "little brother" module.

TerryE, 2015-12-12:
Enjoy your leave.  Don't work too hard on esp8266 stuff :)

TerryE, 2015-12-12:
There are two Espressif RTOS SDKs: [ESP8266 RTOS](/espressif/ESP8266_RTOS_SDK) and [ESP32 RTOS](/espressif/ESP32_RTOS_SDK). Both are based on FreeRTOS (versions V7.5.2 and V8.2.0 respectively).  Free RTOS isn't quite free as in beer because you can't download the documentation free  You need to buy the official manual.  However, I did find [this version](http://thetoolchain.com/datasheets/FreeRTOS_manual.pdf) online from the toolchain.com.

I'll do some background research over the next month or so.  I am not on any beta programme, so I don't expect to touch the H/W soon.

TerryE, 2016-04-11:
I suggest that we close this if and until the esp32 is generally available and we have SDK details.

closed 2016-04-11 by TerryE