nodemcu/nodemcu-firmware #844: Add crypto.toBase64()

TerryE, 2015-12-12:
As it says on the tin.

xvh, 2015-12-12:
It would be nice if you add also crypto.fromBase64. Now I must use base64.dec() function from base64 module in lua modules.

TerryE, 2015-12-13:
Oops got my to and from mixed up.  The to already exists.

TerryE, 2015-12-22:
OK. I've got this working.  I'll push it once we've moved `dev` up to SDK 1.5.

devyte, 2016-02-04:
Hi, I'm wondering whether this will get added soon? I know that stability is the current priority, but this would help with a few things on the lua side.

TerryE, 2016-02-04:
I've got the patch working locally but the consensus was to split out the conversion routines into  separate lib.  Need to finish the GPIO stuff first :sweat:

TerryE, 2016-03-02:
PR #1072

closed 2016-03-02 by TerryE

marcoskirsch, 2016-12-26:
This never made it into the firmware?

TerryE, 2016-12-27:
As far as I recall, it did.

marcelstoer, 2016-12-27:
#1072 introduced a new module `encoder` which contains this. See http://nodemcu.readthedocs.io/en/latest/en/modules/encoder/#encoderfrombase64