nodemcu/nodemcu-firmware #845: gpio.trig doesn't play well with interpreter and can cause a panic

pjsg, 2015-12-12:
The gpio module has support for callbacks which will be triggered when the specified condition happens on the selected pin. The code is called at interrupt level (I think) by the SDK. It can thus interrupt what is going on in the main line code. If you enable the asserts, and then run the following code, you can demonstrate the problem (with a bit of luck):

```
function gp()
    p1 = 0
    du = 0
    gpio.mode(1,gpio.INT)
    function pin1cb(level)
       du = (tmr.now() - p1)
       print(du)
       p1 = tmr.now()
       if level == 1 then gpio.trig(1, "down") else gpio.trig(1, "up") end
    end
    gpio.trig(1, "down",pin1cb)
end

gpio.mode(0, gpio.OUTPUT)
bit = gpio.HIGH
tmr.alarm(0, 100, 1, function() gpio.write(0, bit) bit = gpio.HIGH + gpio.LOW - bit end)
gp()
```

Once running this code, use a screwdriver (or something similar) to make a noisy connection from D0 to D1. Wiggle the screwdriver around, making and breaking the connection. I find that I can trigger assert failures within 10 to 15 seconds, sometimes with panics as here. For example:

```
2718
lapi.c:607: L->top < L->ci->top
lapi.c:447: L->top < L->ci->top
lapi.c:830: (nresults) == (-1) || (L->ci->top - L->top >= (nresults) - (nargs))
PANIC: unprotected error in call to Lua API (stdin:1: attempt to perform arithmetic on field 'HIGH' (a nil value))
```

TerryE, 2015-12-13:
The first is here:

```C
LUA_API void lua_rawgeti (lua_State *L, int idx, int n) {
  StkId o;
  lua_lock(L);
  o = index2adr(L, idx);
  api_check(L, ttistable(o) || ttisrotable(o));
  setobj2s(L, L->top, ttistable(o) ? luaH_getnum(hvalue(o), n) : luaH_getnum_ro(rvalue(o), n))
  api_incr_top(L);   //<<<<<<<<<<<<<<< This fails
  lua_unlock(L);
}
```
and the second here:

```C
LUA_API void lua_pushinteger (lua_State *L, lua_Integer n) {
  lua_lock(L);
  setnvalue(L->top, cast_num(n));
  api_incr_top(L);   //<<<<<<<<<<<<<<< This fails
  lua_unlock(L);
}
```

I'll have a poke around tomorrow.  Well past my bedtime!!

pjsg, 2015-12-13:
It is past my bedtime too! I wouldn't look at those particular failures -- I think that the failure points are pretty random. I think that it comes from the unpredictability of the timing of the gpio interrupt with respect to the main lua code executing.

There is a possibility that these callbacks should be executed with their own Lua_state, but I don't think that it would solve the problem of interrupts just happening at a bad time.

TerryE, 2015-12-14:
OK, I think that I know what _might_ be causing this. It could be one of two issues, but let's run through the timeline.

During the`gpio` module initialisation, the code uses `platform_gpio_init()` to register the cb function `gpio_intr_callback()`, and this in turn  does a `ETS_GPIO_INTR_ATTACH(()` of the cb routine. ***Now I believe that this is a genuine interrupt registration and not an SDK task callback***  This routine is based on some SDK cb examples and scans the GPIO pins which have the trigger enabled and an interrupt pending, and if found it disables further interrupts _for that pin_ and calls `gpio_intr_callback()` with the pin number and level read.  

This was in the **interrupt driver context**, but  `gpio_intr_callback()` runs in the **Lua environment** and hence cb might be interrupting and reentering Lua whilst a lua task is in train.  **This is issue 1**  (@jmattsson, @nodemcu: is this correct?)

Because the cb runs in the interrupt driver context it has not concept of **Lua_state**, so the gpio module does a botch of caching `L` in a local static `gL`.  _However we know that by default, the  GC can relocate the Lua stack after any alloc, resulting in gL pointing into a garbage stack fragment_  **This is issue 2** Note that `net`,`node`,`cjson` and `wifi` as use this practice.

pjsg, 2015-12-14:
My take is that the second problem isn't a problem. I don't think that the Lua_state block itself moves -- so the pointer should continue to be valid. *However*, it is unclear that that approach will work if the user is using coroutines and creating and deleting lua_states.

By all means split these into their own issues -- I thought that I was reporting the first issue that @TerryE  identified above.

TerryE, 2015-12-14:
> I don't think that the Lua_state block itself moves

Sorry but wrong.  This was an issue that we discussed with Nick.  THe stack grows with subroutine depth and the GC can trim it back if the stack isn't fixed. Grep the code for `fixedstack(`

TerryE, 2015-12-15:
@pjsg.  Philip I will do a test patch for this in the next few days, but I won't issue a PR at this stage.  When I do so can you cherry pick this on your local dev setup and try it out with your screwdriver test.  See also   #851.

nickandrew, 2015-12-23:
So how did that screwdriver test go?

Per my comment in #871, it should be possible to hook gpio interrupts to call C functions in other device drivers. Such C functions would operate under interrupt constraints (no calling into Lua, no UART output) but one of those C functions should be to post a Lua task to implement `gpio.trig()`.

TerryE, 2015-12-23:
Yup that's what I implementing but for a 1.5 version.  It should be a 2015 PR :)

TerryE, 2016-01-24:
OK,  There are three major instabilities with the the GPIO code:

-  Stuff that shouldn't be executed out of Flash is being so.  I am talking specifically about the ISR and a couple of interrupt-disabled sections that are in `platform_gpio_mode()`. I pull these last two out into separate functions and tagged all three as `ICACHE_RAM_ATTR` .

-  The ISR can be reentered but is not designed as rentrant

-  It call Lua Land in a nasty nasty way.

The can all be fixed but there still remains an issue of how to handle reentrance.  What I am talking about is say we are triggering on GPIO2 which is bouncing, so the ISR gets a **pin2:close** for which it issues a `task_ post_high()`, but before the task executes pin2 opens and closes again.  If we make the trigger one-shot, the ISR disables interrupts on pin2 and it is left to the Lua cb to reissue the trigger.  If we don't and we have a rapidly bouncing pin, then we will quickly exhaust task Q.  My vote is for one-shot, which is (if I recall correctly) the current intended behaviour).

nickandrew, 2016-01-24:
@TerryE My test in #874 is avoiding Lua land, so your first two points are most interesting to me.

Tagging the ISR functions is easy enough. I had done that in my driver but not for `platform_gpio_intr_dispatcher` which I assume is the main culprit.

I don't understand enough about what `platform_gpio_intr_dispatcher` does to deal with reentrancy issues. Specifically, what's going on here:

```c
      //disable interrupt
      gpio_pin_intr_state_set(GPIO_ID_PIN(pin_num[i]), GPIO_PIN_INTR_DISABLE);
      //clear interrupt status
      GPIO_REG_WRITE(GPIO_STATUS_W1TC_ADDRESS, gpio_status & BIT(pin_num[i]));
```

At the moment I'm just following the discussions, waiting for some code I can test.

TerryE, 2016-01-24:
The _current_ ISR still calls  `gpio_intr_callback()` even if the pin doesn't have a trigger on it.  Given that this is all running from Flash, this is probably breaking the 50 µSec rule.  ISRs should be lean and mean and run from RAM.

pjsg, 2016-01-24:
My approach to the GPIO interrupts is as follows:

* Individual events on a pin can be lost, but at least one will be delivered. The triggers are not single-shot, they last until cancelled.

I.e. if you have a trigger on falling edge (say on a button), and someone mashes the button, then you should get at least one callback, but no more than the number of falling edges. 

Also, there is no particular guarantee that events are delivered in the order in which they happen. 

The ISR does:

* Get the triggered bits (i.e. read the status register)
* Acknowledge those bits into the W1CT register
* Or this into 'pending triggers'
* Check 'low-level-task-queued'
* If not set, then post a message with the 'pending triggers value'
* If the post works, then set low-level-task-queued and clear 'pending triggers'
* return

The low level task does
* clear low-level-task-queued
* Call the callback for each supplied 'pending triggers' value (that was passed via the message)
* disable gpio interrupts
* if not low-level-task-queued:
    grab 'pending triggers' into temp
    set 'pending triggers' to 0
* reenable gpio interrupts
* If temp is non-zero, restart this handler and call all the required callbacks.

This approach (I think) deals with every case except when the ISR gets an interrupt and fails to post the message as the queue is full, and there is no pending low-level task. In this case, the trigger will be delivered when the next event happens. The only fix to this is to have a timer that deals with the task queue overflow case.

This approach only has a single GPIO entry in the task queue at a time. If we require that each use of the task mechanism also obeys this, then there is a natural limit on the number of entries that can be in the queue.

I'm assuming that the GPIO ISR cannot be interrupted by another invocation of the same ISR -- but this is pretty standard.

TheoHarbers, 2016-01-25:
Just an idea: why not capture all gpio states when one triggers together with a microsecond timestamp/counter and put it in a table/queue. if the queue is full remove oldest entry or loose last input. If this is fast enough it can all be inside the ISR. Call one callback if not already busy. In the callback read the queue in a loop until empty and dispatch from there what ever the user want to do with it. This may allow also to decode pulse widht modulated signals IR/RF decoding.

pjsg, 2016-01-25:
I looked at examples of gpio.trig that were found online: https://www.google.com/search?q=gpio+trig+filetype%3Alua&rlz=1C1CHFX_enUS563US563&oq=gpio+trig+filetype%3Alua&aqs=chrome..69i57.3959j0j7&sourceid=chrome&es_sm=93&ie=UTF-8

Apart from the fatc that there weren't very many, I observed:

* One person thought you had to redo the trigger after each callback (i.e. oneshot)
* One person thought that the trigger was persistent
* One person assumed that the trigger was called reasonably quickly as they used the time to decide what to do.

If nothing else, it shows that either people don't read the documentation, or that it is unclear!

I would favour adding the time of the event to the callback arguments. This would enable lots of stuff. I don't know how deep the queue would have to be in order to not drop stuff. I fear it might have to be at least a hundred slots deep. I wouldn't make these task queue entries, but just do this queue inside the gpio module.

TerryE, 2016-01-26:
Philps my next response will be onn #874

TerryE, 2016-02-06:
@pjsg  @nickandrew @jmattsson to name but a few, adn in response to Phillip's last post, the current implementation of trig is persistent -- unless you set the mode of the triggered pin to other than gpio.INT or replace the trigger.  I've been playing with a sketch that I wrote for Marcel to investigate the behaviour of the current triggering implementation.  This minimises the time in the Lua callback by only storing the value in an array, which is then formatted and dumped after a fixed dwell.
```Lua
local pin = 4    --> GPIO2

local function debounced_trigger(pin, onchange_function, trigger_type, dwell)
  local start = tmr.now()
  local times = {}  -- collect trigger time offsets +/- = high/low
  local last_event
  local now, alarm, mode, INPUT = tmr.now, tmr.alarm, gpio.mode, gpio.INPUT
  local function trigger_cb(event)
    local delta = now() - start
    if not last_event then alarm(6, dwell, 0, function()
        -- note that pin and last_event are upvals with values as to function firing
        mode(pin,INPUT)
        onchange_function(pin, last_event, times)
      end)
    end
    last_event = event
    times[#times+1] = (event == gpio.HIGH) and delta or -delta
  end
  gpio.mode(pin, gpio.INT)
  gpio.trig(pin, trigger_type, trigger_cb)
end

local function onChange (pin, last_event, times)
  local format = string.format
  print(('The pin value has changed to %d with last event %d after these contacts:\n'):format(
       gpio.read(pin), last_event))
  for i=1,#times do 
    local level = times[i] > 0 and "high" or "low\n"
    local t = times[i] > 0 and times[i]  or -times[i]
    times[i]=format("%8.4f:%s", t/1000000, level)
  end          
  print(table.concat(times, ''))
end

debounced_trigger(pin, onChange, 'both', 5000)
```
The nature of the response depends very much on the cleanness of the physical contact and the trigger type.  Any level rather than edge just overloads the memory.  Here is a typical data set:
```
> The pin value has changed to 0 with last event 1 after these contacts:

  1.6017:high  1.6052:low
  1.7351:low
  3.1606:high  3.1642:low
  3.1653:high  3.3700:high  3.3927:high  3.4004:high  3.4010:high  3.6702:high  3.9980:low
  4.6258:high  4.6276:high  4.6395:high  4.6399:high  4.6965:low
  4.6969:high  4.7077:high  4.7534:high
```
As I said, the current code will deliver callbacks as fast as it can, which in this case for a high trigger might be 5+ per mSec. What I am proposing for the new task based version is:

-  We keep the triggers persistent, but
-  We only reenable the trigger as part of callback delivery.  If I don't do this then all that happens is that the ISR floods the task queue with repeated callbacks.

nickandrew, 2016-02-07:
@TerryE Are you proposing to convert everything in the `debounced_trigger` function into C? That's where it belongs. I'm not sure if C can manipulate the `times` array from within an ISR.

I like the idea of delivering the trigger to Lua through a timer callback (at least) 5ms after the interrupt occurred. That's a neat way of separating Lua from the ISR function.

TerryE, 2016-02-07:
@nickandrew, No this stuff is definitely Lua-land.  I apply a sort of "Render unto Caesar" argument: some things are better done in C and some in Lua.  This code is just a test harness which allows you to play with the interrupts.  What really surprises me is the electrical characteristics of simple brush contacts which some form of RC or more active deadbanding.  I could really do with having a look at these signals with a scope.  

I've got the latest iteration of the gpio code working stably and integrated with the task interface.  I just need to finish some of the documentation before doing the PR..

TerryE, 2016-02-08:
OK all, please see #1023. This stuff is so difficult to test that this is a minimal functional change, and I've only changed the core tasking and gpio support.  I don't see this ready to pull into dev, but it should still be a _lot_ more stable than dev as it currently stands.

TerryE, 2016-03-02:
I think that with the new tasking I/F and GPIO rework, we've now covered off all of this.

closed 2016-03-02 by TerryE