nodemcu/nodemcu-firmware #846: CJSON is incompatable with Lua coroutining

TerryE, 2015-12-13:
I've just been making some blanket changes to the modules and noticed that CJSON assumes a single global state in 

```C
static lua_State *gL;
\\ ... 
void cjson_mem_setlua (lua_State *L)
{
  gL = L;
}

void *cjson_mem_malloc (uint32_t sz)
{
  void *p = (void*)c_malloc (sz);
  if (!p && gL)
// ...
```
Separate coroutine threads have separate stave vectors L.  This code assumes that there is a single global state vector.  Not good.

nickandrew, 2015-12-13:
@TerryE Any idea why it does it? To save memory?

TerryE, 2015-12-13:
No because it is a plug in replacement for c_malloc and malloc doesn't accept the state.  PHP has the same concept as L for threaded code (called TSRM) , but it provides a macro which is called `TSRMLS_FETCH()` to recover the current thread vector if you are in a routine which doesn't have it  passed down.  This does some magic with the stack to work out which thread is running.  We could do something similar, but the easiest way is to pass L into the allocator by defining `CSON_MALLOC(s)` as `cjson_mem_malloc (sz, L)

TerryE, 2015-12-14:
As I discuss in #845,  `gpio`,  `net`, `node`,`cjson` and `wifi` all use this trick.  It can fail if the application uses coroutines because each thread has its own `Lua_state`, and applications like the http server example uses coroutines.  

**eLua** has the lua function `lua_getstate(()` which always returns the main thread, so we could use this instead, but the issues arise if one thread executes the API call which sets `gL` but another thread executes the call which accesses it, because the wrong thread context will be applied, so for example upvalues will be garbage.  Likewise if the setting routine exits, its thread is freed so gL will pointing at garbage. 

This one could be a minefield so I need to think this through.

TerryE, 2015-12-14:
OK, one aspect that I did get wrong when I've checked: both globals **_G** and the Lua registry are shared across threads:

```
> =debug.getregistry()
table: 0x3fff0a10
> x=coroutine.create(function()print(debug.getregistry())end)
> =x
thread: 0x3fff1b68
> coroutine.resume(x)
table: 0x3fff0a10
> 
```
(Both standard Lua and nodeMCU Lua on the ESP show this behaviour.)  

Thinking about this, I think that we are OK always to use the main thread for this type of context so we should just convert all of these to `lua_getstate()` .  Any alternative opinions?

pjsg, 2015-12-14:
I think that callbacks should be executed on the same lua_state as when they were registered. Otherwise I suspect that the upvalues will be wrong (for some definition of wrong).

Alternatively, they should be executed in a new lua_state. I do think that the approach should be consistent between all the different callbacks. For example, what ought to happen here:

```
local pin = 0
function logtrig(level)
  print(pin, level)
end

pin = 1
gpio.trig(1, "both", loglevel)
pin = 2
gpio.trig(2, "both", loglevel)
pin = 3
```

When pin 1 fires, *should* it print out the pin value of 0 or 1 or 2 or 3?

TerryE, 2015-12-14:
Philip, Your example is more to do with standard Lua and an aspect that confuses many new to Lua programmers. You have to differentiate between three different epochs: when the function is compiled; when it is bound to a closure and when that closure is executed.  A function is compiled when the file containing it is loaded.  It is bound at runtime to a closure when the execution passes the point in the code where it is declared.  It is executed when it is called.

So in your example the local pin = 0 when the closure is bound so if pin 2 went high then 1 went low, you would have output:

```
0	1
0	0
```
The following would achieve that I think that you are intending:

```
for pin = 1,2 do
  gpio.trig(pin, "both", function(level) print(pin, level) end)
end
```
Here pin is a loop local variable and the two calls to `gpio.trig()` would bind the correct values:

```
2	1
1	0
```
---------
On a separate point I've been thinking in terms of the Lua architecture how coroutining impacts and interacts with the SDK.  The SDK essentially treats the init.lua, and and any UART / `node.input()` as SDK tasks and (normal) callbacks as atomic as it is non pre-emptive (other than ISRs discussed in #851), and so the SDK can only regain control to execute the next task / callack once the currently executing one has returned control.  This can only be done by return from a main thread task and any coroutines are suspended having yielded control to the main thread.

pjsg, 2015-12-14:
My bad!

I was trying to come up with an example where there would be a difference between the callbacks happening on the main task lua_state, and where the registration of the callback would create a new lua_state for that registration (i.e. one per pin). Another option is that the registration is tied to the current lua_state -- i.e. if you did a registration out of a coroutine, then you would get the state of the coroutine rather than the main task state. 

It seems that these three options would have observable differences. 

I think that I am somewhat confused....

TerryE, 2015-12-15:
Not really.  It's just bloody counter intuitive.  It took me a while to get my head around it.  If you think of it, they main thread is the only one that  can interact with the SDK tasking system.  So if you did do a `gpio.trig()` in another coroutine thread, then the cb task has to "fire" in the main thread, so cb's should always bind to this.

jmattsson, 2015-12-15:
I'm the culprit who added the cjson Lua mem wrapper stuff. The main purpose was to get a lua_State\* available whenever the cjson lib does an allocation (which happens N levels deep through various functions) and be able to raise a Lua error on a failed allocation (since the cjson lib is erm... very optimistic about memory availability). Actually passing an `L` through the call chain all the way down to the actual alloc function would have been a huge job and left us with something that wouldn't have been possible to upgrade from upstream without another huge effort.

At the time I went through the source and I *thought* there was only ever one lua_State\* allocated during startup (and then never even realloc'd, only items "inside" it). @TerryE are you sure the actual "main" `L` pointer can ever change? Because I was sure it didn't, so I only felt a little bit dirty adding the wrapper the way I did (which btw I will still claim is superior to the previous behaviour which was a system-wide crash).

I admit that I never even considered coroutines since I haven't made any use of them myself, but it makes sense that they might have their own lua_State\*.

As for fixing this particular instance (being the cjson module), I guess the cleanest (if it can even be called that) would be to call `cjson_mem_setlua()` first whenever any module function calls into the cjson lib. Given run-to-completion that should be safe, and work with multiple lua_State\*s.

Of course, if someone was to actually teach the cjson lib that memory allocations *CAN FAIL*, that would be amazing.

TerryE, 2015-12-15:
@jmattsson Johny, it took me a bit of thinking to get my head around this.  As I discussed above with Philip, the stack associated with a thread state _can_ move but the main thread Lua_state, let's call it **L0**, is allocated at Lua startup and never freed.  There are sound reasons why we should always use **L0** for this type of application because:  

1.  It always exists
2.  Even if the call was from a coroutine thread, say **L1**, this means that **L0** is in _resume_ state so is in a sensible condition.
3.  We will only pass information down the stack and so long as the calling function ensures that any called Lua code hasn't left any return values on the stack, then there is no issue with using the wrong state.

The only issue here is that these called routines might have resulted in the **L0** stack growing above the realloc threshold or triggered a stack trim realloc so we need to make sure that the resume / yield to return code doesn't make assumptions about the stack not having be reallocated.

So once we've done this check, my recommendation is that we dump all of the local gL caching and just replace this everywhere by a call to `lua_getstate()` which always returns **L0**.

I'll put in  a bulk PR to implement this once the LTR cleanup has been merged.

jmattsson, 2015-12-15:
What about upvalues inside callbacks registered from co-routines? Would that still be bound/available correctly?

TerryE, 2015-12-15:
OK, the td;dr answer to this one is that they are, but why is really complicated.  And for the detailed answer.  

The two essential references that you need are [A No Frills Introduction to Lua 5.1 VM Instructions](http://luaforge.net/docman/83/98/ANoFrillsIntroToLua51VMInstructions.pdf) and Roberto Ierusalimschy's paper [Closures in Lua](http://www.cs.tufts.edu/~nr/cs257/archive/roberto-ierusalimschy/closures-draft.pdf).  So consider the following and run it through a `luac -l`:

```Lua
function g() for i=1,3 do collectgarbage() end print(node.heap()) end; g()
do local X={'fred'} 
   function pushX(v) X[#X+1] = v end 
   function popX() local v=X[#X] X[#X]=nil return v end
   X={'bob'} print (popX()) 
end 
g() for i=1,100 do pushX(i) end;  g()
pushX, popX = nil, nil; g()
```
It doesn't matter whether pushX and popX are globals or themselves upvalues; I want to discuss what happens to X.  When the source is compile a Proto structure is created for each compiled do block and functtion, but this does nothing other than contain the R/O parts of the compilation O/P.  It isn't executable. It becomes an executable object, a closure, when the execution passes through the instruction sequence where it was defined and the VM executes a `CLOSURE` plus N (in this case 1) `MOVE` / `GETUPVAL` instructions.  These last aren't really instructions but a mechanism to extend the closure instruction.  These map the upvalues to locations in the stack.  So in this case the pointer to X gets initially stored in `CL->upvals[0]`  (search `lvm.c`for upvals to see where these are used).  

So pushX and popX both have a shared Lua `LUA_TUPVAL` object which references the do local X _not_ the table which X contains, which is why the first `print (popX()) ` returns "bob" not "fred".  At this stage the upvalue is referenced  and is therefore not GC collectable.  This is known as an open upvalue.  

But what happens when the do `end` is executed as the locals are discarded?  This is where the Lua magic happens.  Follow through from `OP_RETURN` to `luaF_close()`  Each upvalue is converted into a local (closed) copy and its reference is added to the G(L)->rootgc linked list.  This is scanned in the normal GC sweeps but its elements are marked as referenced if used in any upval list of any marked functions  This prevents it from being GCed whilst any functions exist that reference it. 

In the above example, when the push and pop functions are dereferenced when the pushX and popX are both set to nil. and at this point the table can be CGed.

Lastly note that whilst L is thread-specific G(L) is the same across all threads.

jmattsson, 2015-12-15:
Thanks for taking the time to explain that Terry!

TerryE, 2015-12-15:
To be honest Johny, I assumed that you were probably familiar with a lot of this, but I suspect that this issue might get a wider readership :)

pjsg, 2015-12-15:
I'm slightly worried about error handling -- is there any interference between errors happening in a callback and any pcall that is open on __L0__? Or are all the callbacks wrapped in an effective pcall themselves? Or is __L0__ *guaranteed* to be empty/idle when a callback fires?

Anyway -- thanks for taking the time to explain this -- I must confess to being fascinated by the minutiae of embedded systems!

TerryE, 2015-12-15:
Philip,  consider how  the \*nix bash shell uses executable commands as primitive function.  If we leave aside the detach operator, the shell only regains control once the exec has returned it.  The shell can execute its logic to decide what exe to run next.  The SDK interacts with tasks / cb's (I'll call them collectively _tasks_) much in the same way.  A task can only run to completion; it's the SDK which decided how to sequence them.  That's why my example in the FAQ `node.restart(); for i = 1, 20 do print("not quite yet -- ",i); end` prints out twenty "not quite yet" before executing the restart task.  

*   The **L0** thread is used to execute all tasks.  When they return control to the SDK, **L0** still persists, though by this time with an empty stack.  
*   Coroutines are really just a syntactic way to invert a separate logical thread into this framework. Each has its  state is stored in its own  a separate State **Ln** with its own stack.  The coroutine has to yield control back to the **L0** task for it to complete.
*   So yes at this stage **L0** is idle / empty and an **Ln** in a paused state, when any new task is initiated (ISRs aside).
*   The pcall mechanism is simply a way of calling a function, but still controlling and intercepting a lower level throw in the current call hierarchy.  In the case of a coroutine thread uncaught calls just die silently.  This is just standard Lua behaviour.  Try the following on the PC and in nodemcu; then swap the `..` operator for the `+` operator.

```Lua
c, o=coroutine, coroutine.create(function() print ("a") c.yield() print ("a"..(0)) end)
for i=1,3 do if c.status(o)=="suspended" then c.resume(o) else print(c.status(o)) end end
```

TerryE, 2016-02-24:
There are four routines which currently declare and use a static `gL` state:
```bash
grep -PIRc "\bgL\b" app | grep -P "\.c:[^0]"
app/modules/net.c:55
app/modules/uart.c:6
app/modules/wifi.c:14
app/cjson/cjson_mem.c:6
```
In addition the following use some other table storage to pass the lua_State context from the callback booking thread to the callback thread, and for the reasons discussed in #1081, this is definitely the wrong thing to do.  Callbacks should always use L0.

-  `sntp.c` uses the callback argument to pass back current Ln
-  `coap.c` used a context structure `lcoap_userdata` field much as `tmr`.
- `mqtt.c` ditto with `lmqtt_userdata`
-  `wifi.c` as per `sntp.c`

jmattsson, 2016-02-24:
*Callbacks should always use L0.*

Is it really that clear-cut though? What if the callback has upvalues in Ln? Is it not safer to issue callbacks with the context they were registered from?

pjsg, 2016-02-25:
I was trying to construct an example where it made a difference which stack the callback was executed on, and *I* couldn't. If someone can actually produce an example where it makes a difference, then I think that we can have the discussion as to what is the right solution.

TerryE, 2016-02-25:
>> Callbacks should always use L0.

> Is it really that clear-cut though?

Unfortunately, yes.  Think about it: the SDK calls a C  task or callback, which should then return control to the SDK within ~15 mSec, which this top level routine does that regaining control from any called subroutines and `return`ing. The C routine may call a Lua function using a lua call, but to regain control that Lua function must first return control by issuing a `return` or running out to its `end`.  

If a function is called as a coroutine, then yes it may yield and be resumed, but when it finally returns its thread is placed in the `dead` state.  So to summarise:

-  The SDK calls C wrappers which call Lua functions in *L0* which may exit to return control to C then the SDK.

-  Lua functions may initiate a new thread *Ln* through the coroutine.create() function which yield / resume from the initiating thread (usually *L0*) but when they finally return they die.

- If the coroutine object is stored in a persistent variable (e.g. a global or a closed upval) and has yielded to is parent *L0* function which then completes, then it can be resumed by a subsequent task running out of *L0*. 

These rules work fine.  You can use the *L1* thread to implement procedural control, but the foreground routines called by the SDK all run in *L0*.

Now in principle we might be able to use the SDK as the L0 thread and get it to call a `coroutine.create()` a new *L1* thread and somehow pick up control again when the *L1* yields and return to the SDK so that the next time it executes then it does a resume.  But this is all complicated and untested, and remember that we have no control over the order in which tasks get delivered.  If a coroutine yields it _must_ resume at the next statement.  It can't jump to another (callback function).  My head just hurts trying to get this usecase workable.

> Is it not safer to issue callbacks with the context they were registered from?

Not really.  Threads can already share globals and both open and closed upvals,  The glosed upvals list is maintained in the `global_State` field `uvhead` so this is all tried and tested.

dnc40085, 2016-02-25:
First of all, I accept the task of fixing the lua_state issues in both the `tmr` and the `wifi` modules.

@TerryE I made the changes you suggested to the `tmr` module and here is the [result](https://github.com/dnc40085/nodemcu-firmware/commit/92ea3995c4b0eb235b2765b737acfa59772bb1e8).

dnc40085, 2016-02-25:
@TerryE [here's](https://github.com/dnc40085/nodemcu-firmware/commit/8e8861227648a893297c1b8a47bb5fd50764c79f) the updated version.

dnc40085, 2016-02-25:
@TerryE [here](https://github.com/dnc40085/nodemcu-firmware/commit/276a0f4941186ea00c8c32cc844ca614c6d57226) are the changes to `wifi` module.

TerryE, 2016-03-17:
Just doing a check on another issue and I note that `net`, `wifi`, `coap`,`uart` and `mqtt` still contain coroutine incompatible usage of saving the calling `L` for callbacks. I've just prepared a PR for all bar MQTT.  I'll leave that one to @pjsg Philip, since he's currently working on MQTT anyway.

Once these have been merged, I'll close this issue

PR #1423 (2016-08-02): Update modules to be Lua coroutine compatible
Fixes #846.
Recreation of #1167 with resolved conflicts.
closed 2016-08-02 by marcelstoer