nodemcu/nodemcu-firmware #847: IS the COAP message ID supposed to be seeded or not?

TerryE, 2015-12-13:


TerryE, 2015-12-22:
The truth of the matter is that COAP is moribund through lack of a maintainer.  All of the COAP bugs are really just a fact of this.  My recommendation is that unless we can find someone who is willing to step up here, we should move this formally to a `ǹot supported` folder and withdraw it as an option for 1.5 if and until we can find a volunteer to sort this.  We can also close all coap issues.. Anyone picking this up can search the closed issues and reopen any that are felt to be worth tracking.

jmattsson, 2015-12-23:
A quick change from `rand()` to `os_random()` would fix this particular issue. Assuming that rand isn't already going through to os_random (which is a hardware (p?)rng) - I haven't checked.

TerryE, 2015-12-23:
Yup but there are other open issues pending for months and one is to the effect that COAP as implemented is unusable.  As I said IMO its moribund and I don't have the bandwidth to fix it, do you?

marcelstoer, 2016-07-14:
A fix for one-of-several issues is still better than none at all I'd argue -> #1401.

PR #1401 (2016-07-17): Seed CoAP message id, fixes #847
Fixes #847.

- [x] This PR is compliant with the [contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [ ] The code changes are reflected in the documentation at `docs/en/*`.

Committers supporting this PR: leave blank
commit cc2fee5e (2016-07-17): Seed CoAP message id, fixes #847 (#1401)
closed 2016-07-17 by marcelstoer