nodemcu/nodemcu-firmware #848: Add node.rst_info()

nickandrew, 2015-12-13:
This *doesn't work* because it always returns reason code 6 "external system reset", but it would be good if it can be fixed.

I noticed the SDK implements system_get_rst_info() which I gather returns the same information as the message which sometimes appears after an exception:

```
Fatal exception 28(LoadProhibitedCause):
epc1=0x402566d5, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00004655, depc=0x00000000
```

It would be good to capture this and make it available to lua code. So I implemented it in my branch here: https://github.com/nickandrew/nodemcu-firmware/commit/dccd9ca4cd6bfb3aa578610c352502eab4fc3966

But it doesn't work. Reading comments here http://bbs.espressif.com/viewtopic.php?t=1230 and here https://github.com/esp8266/Arduino/issues/1098 indicates it is fixed in SDK 1.4.0 or fixed in 1.4.0+patch or fixed in 1.5.0

I haven't tried the patch yet but perhaps we can leave this issue open for testing pending 1.5.0 SDK support (add a milestone?).

devsaurus, 2015-12-13:
Interestingly, this function seems to provide more information than `rtc_get_reset_reason()` which is currently [exposed as node.bootreason()](https://github.com/nodemcu/nodemcu-firmware/blob/7512e628b6b39a8ab4f85e73c13ca216a1734eaa/app/modules/node.c#L474). Both should be merged / unified once `system_get_rst_info()` can be considered working.

TerryE, 2015-12-13:
@nickandrew Nick I think that your patch is making this implementation overly complicated.  

*    The ROM / reboot code dumps a 7 word info block into the RTC RAM.
*    The SDK copies this into a 7 word RAM structure during its startup and makes the address of this available through the `system_get_rst_info()` API call.  There is no point in the firmware taking a second RAM copy.
*   As Arnim says we already have a node. bootreason() call, so why not simply extend this to return the extended list.  This has the advantage of reusing the exting firmware API call in an upwardly compatible way.
*    I dislike system calls dumping info to the uart as a side effect.  If they print then that should be the function and the call should be `node.printbootreason()` or whatever, but what's the point of this if you can simply do:

``Lua
  print( ('reset reason: %x exception: %d epc1=0x%08x, epc2=0x%08x, epc3=0x%08x, excvaddr=0x%08x, depc=0x%08x'
       ):format(node.bootreason()) )
```

Anyway here is the version of bootreason that does this :smiley:

```C
static int node_bootreason (lua_State *L)
{
  const uint32 n = (sizeof(struct rst_info) / sizeof(uint32));
  uint32 i;
  uint32 *restart_info = (uint32 *) system_get_rst_info();
  for (i = 0; i<n; i++) lua_pushinteger(L, *restart_info++);
  return n;
}
```

nickandrew, 2015-12-13:
Hi Terry,

That was just proof of concept code, which is why it was doing all kinds of ugly things. I'm surprised that you got actual results out of it, cause all I ever got was reason code 6.

TerryE, 2015-12-13:
Mine uglier, but shorter :)

nickandrew, 2015-12-14:
Which SDK variant did you use?

TerryE, 2015-12-14:
SDK 1.4 -- the latest dev build.

nickandrew, 2015-12-23:
I tried your code but I am still getting only reason 6 after forcing a WD timeout and restart.

I then tried replacing bits of the SDK as espressif advised on 16 Oct (in one of the links above):

```
$ unzip -v system_get_rst_info_patch.zip
Archive:  system_get_rst_info_patch.zip
 Length   Method    Size  Cmpr    Date    Time   CRC-32   Name
--------  ------  ------- ---- ---------- ----- --------  ----
  187258  Defl:N    62059  67% 2015-10-16 09:54 928f4876  libmain.a
  220958  Defl:N    80889  63% 2015-10-16 09:54 38558f45  libpp.a
--------          -------  ---                            -------
  408216           142948  65%                            2 files
```

That didn't compile though as there is now a function naming conflict with pwm_start and friends now appearing in `libmain.a` when they never did before. So I don't know what Espressif did with this "patch" as it's clearly not minimal.

The original versions of those .a files which I am pretty sure are being linked in are:

```
$ md5sum ./sdk/esp_iot_sdk_v1.4.0/lib/libmain.a ./sdk/esp_iot_sdk_v1.4.0/lib/libpp.a
6f13fa17af0195fd9f96e11a0c148d8f  ./sdk/esp_iot_sdk_v1.4.0/lib/libmain.a
f72b7dd6a35a43e7c67a3b7b94dbd47e  ./sdk/esp_iot_sdk_v1.4.0/lib/libpp.a
```

So ... no idea. It would have been good to debug how GPIO interrupts are causing crashes even without Lua's involvement.

TerryE, 2015-12-23:
@nickandrew Nick, ONe thing that I've learnt when supporting tihs type of complex systems is to look forward.  We are fixing bugs.  We are also rebaselining against the SDK 1.5 which also has its own bugfixes.  So I've explained in other posts, what I want to do is to rebaseline to V1.5 first and implement the fixes  against this baseline.  Let's do the work and discuss it.  ttrying to reconcile against 1.4 is a distraction.

nickandrew, 2015-12-23:
No worries, this can be tested later with 1.5 as it's supposed to be fixed (again) anyway.

closed 2016-01-15 by TerryE