nodemcu/nodemcu-firmware #859: unreliable ds18b20 readings

eyaleb, 2015-12-18:
I had some issues in this area and now decided to look closer and reread the datasheet. I believe that our module has a problem.

What we do now is this 
```
  ow.reset(pin)
  ow.select(pin, addr)
  ow.write(pin, 0x44, 1)  -- start a new conversion

  ow.reset(pin)
  ow.select(pin, addr)
  ow.write(pin,0xBE,1)  -- read the temperature
```
What is missing is the waiting for the conversion to complete. What the above does is read the old value while it is being updated. It regularly returns lower bits error.

The correct way is something like this
```
  ow.reset(pin)
  ow.select(pin, addr)
  ow.write(pin, 0x44, 1)
  if not done_conversion() then return nil end
  ow.reset(pin)
  ow.select(pin, addr)
  ow.write(pin,0xBE,1)
```
The wait looks like this
```
local function done_conversion()
        local t = tmr.now()
        repeat
                local status = ow.read(pin)
                if tmr.now()-t > 750000 then    -- 750ms limit
--                      print ("convert timeout")
                        return false
                end
                tmr.wdclr()
        until (status ~= 0)
--      print ("converted in "..((tmr.now()-t)/1000).."ms")
        return true
end
```
However, this takes 600ms (measured on my esp-12) which is unacceptable to me - my app runs in under 250ms. So what I do now is simple and reliable - reverse the order of the operations. Since I only read once every 1 or 5 minutes this suits me. And this is what was happening before so this is not a changed semantics.
```
  ow.reset(pin)
  ow.select(pin, addr)
  ow.write(pin,0xBE,1)    -- get earlier reading

  ow.reset(pin)
  ow.select(pin, addr)
  ow.write(pin, 0x44, 1)  -- start a new conversion
```

nickandrew, 2015-12-18:
One man's unacceptable delay of 600 ms conflicts with another man's unacceptable getting temperature reports that are 1 to 5 minutes old.

In my code I just setup a timer to callback into the processing function after 1 second.

In the datasheet there's a documented way to query all devices on the bus to wait until they have all completed conversion, but I think that's just overkill compared to waiting 1 second.

eyaleb, 2015-12-18:
@nickandrew It is a given that requirements vary. For me adding 600ms to what now takes 300ms means the batteries will last only one third of the time. Not nice.

But to the main point: right now there is no "wait for the conversion to complete" so everyone gets the old reading, at best.

Maybe we need a function to get the old reading and another to make a fresh one.

eyaleb, 2015-12-19:
I investigated further and discovered that another change (of mine, see https://github.com/nodemcu/nodemcu-firmware/pull/746) was the main reason for the bad reading. A good reading is made regardless of the order of the two operations (read, convert).

Nevertheless, either way the reading is an old value and not a current one. To get a new reading one must start a conversion and wait for the result (see `The correct way...` above).

TerryE, 2015-12-19:
Whatever the approach you shouldn't do a computer loop watching the timer for more than 1 msec or so. At minimum you should use a one-shot tmr.alarm or even  sleep on low powered applications. After all the time is needed by the ADC within the ds18 chip, not a CPU process.

eyaleb, 2015-12-19:
@TerryE I do not know how to use a timer inside the ds18b20.lua module which needs to return a reading.

The only way I see is to have a separate "start converting", "is convert finished?" and "fetch reading" functions. Then my app can sleep part of the time. Right now I read, check for a failure then repeat (all on a timer). But the ds18 does not return a failure is the conversion is not finished (or so I see).

However, if I can read the old temp (no waiting required) then start a conversion and go to sleep for 1 or 5 minutes, then it is best. I thought that this is what is happening but I have some issues with bad readings that I did not get to the bottom of yet. The ds18 stays powered, so I expected it will chug along converting while the esp sleeps, having a new reading in time for the next wakeup. This mostly works but not always.

I am not giving up just yet.

TerryE, 2015-12-19:
Well I have a bunch of ds18b20s that I need to wire up at some point, so I am very interested in your experiences :smile:

eyaleb, 2015-12-19:
Later: one esp-12 that was misbehaving ran well overnight after reseating some of the wires. I should stop using breadboards (or get better quality ones?). I have soldered many of my esps on vero-boards and they work well so I need to do the rest.

TerryE, 2015-12-19:
@eyelab, this is _really_ outside the scope of this list, but a quick answer.  I use double sided through-hole boards and 40 SWG for tracking or solder bridging for adjacent holes. Far neater than Veroboard, IMO.  I also power mine using a cheap 5V -> 3.3V module and a A type female USB connect so that I can just stick my ESP module into one of these cheap 5V USB charger plugs and so that I can plug it it any socket in the house.  Just don't forget the capacitors.

closed 2015-12-19 by TerryE