nodemcu/nodemcu-firmware #860: dofile() doesn't works if file name become from variable

pastukhov, 2015-12-18:
Hi!
I'm trying to make some dofile() wraper : 
```lua
function dolc(filename,data)
    if  file.open(filename..'.lua') then
        file.close()
        print('Compiling:', filename..'.lua')
        node.compile(filename..'.lua')
        file.remove(filename..'.lua')
        dofile(filename .. '.lc')(data)
    elseif file.open(filename..'.lc') then
        file.close()
        dofile(filename .. '.lc')(data)
    else
        print('Can`t find '..filename.. ' to exec')
    end
end
```
But i'm getting 
```
init.lua:10: attempt to call a nil value
stack traceback:
	init.lua:10: in function 'dolc'
	init.lua:32: in main chunk
	[C]: in function 'dofile'
	stdin:1: in main chunk
```
Line 10 is ```dofile(filename .. '.lc')(data)```
and 32 : ```dolc('splash')```
Is it dofile() bug or my code problem?

pastukhov, 2015-12-18:
Sorry, i fond the mistake.

closed 2015-12-18 by pastukhov

TerryE, 2015-12-18:
Atrem,  always pcall compiles as the compile can have errors.  
Also remember that it's safer to flush after file operations, especially any removes()
Also file.list() returns a table  containing the file directory.  It's quite a lightweight operation so `files.list()[x]`is a cheap way to check if `x` exists.  Try:
```Lua
print(file.open('fred','w')) file.write( 'hello') file.flush() file.close()
print(file.list()['fred'])
print(file.list()['notfred'])
```
to see what I mean.