nodemcu/nodemcu-firmware #862: Reset on Interrupt during a require

fatshady, 2015-12-19:
In my project i have an optical encoder. I read the A and B channels by using an Interrupt using 
gpio.mode(chA, gpio.INT)
gpio.trig(chA, "up", doEncoder)
i can count the steps correctly and it's fine.
To save ram i load the function in memory only when i need them using a "require"

i have a timer that every 2 second do a control...in the function called by the timer i noticed that if the module recive an interrupt while is doing a "require" it reset.

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 26636, room 16 
tail 12
chksum 0x1f
ho 0 tail 12 room 4
load 0x3ffe8000, len 2264, room 12 
tail 12
chksum 0x3f
ho 0 tail 12 room 4
load 0x3ffe88d8, len 8, room 12 
tail 8
chksum 0xe2
csum 0xe2
don't use rtc mem data
rlŽ‚rlŒžâŒblŒbþ

im also wondering what "don't use rtc mem data" means.

TerryE, 2015-12-19:
Sorry but this issue falls under #719, so please read this link.  This is the wrong place to post this type of question.

closed 2015-12-19 by TerryE

fatshady, 2015-12-19:
Sorry i thought maybe this issue were linked to #851.

TerryE, 2015-12-19:
@fatshady, no need to apologise.  We will help out with constructive comments if we can, but our priority on _this_ list is to work development issues.  The other forums do application programmer support.  There are some 101s here. like make sure that you only load the modules that yuo need and make sure that you properly overlay your code so that each phase is ephemeral, but if I start discussing this stuff, then we will just get snowed undr with Q that are best placed on SO or esp9266.com.

jmattsson, 2015-12-20:
> im also wondering what "don't use rtc mem data" means.

I've answered this [before](https://github.com/nodemcu/nodemcu-firmware/issues/302#issuecomment-151401132), as a search for the "don't use rtc mem" here would show.

fatshady, 2015-12-21:
@TerryE ok thank you, i understand.
@jmattsson i searched for days but i didn't find anything... thank you very much, very usefull