nodemcu/nodemcu-firmware #864: Dev branch no WiFi Connection.

confile, 2015-12-20:
I tried to run this lua script: https://gist.github.com/confile/1ccf0783a2a25a585f72

I created a dev firmware with all 31 modues, with SSL on, no debugging: Here is this firmware Creation date: 2015-12-20: [nodemcu-dev-31-modules-2015-12-20-15-28-14-float.bin](https://dl.dropboxusercontent.com/u/47067729/nodemcu-dev-31-modules-2015-12-20-15-28-14-float.bin).

Using ESplorer to upload the init.lua script I get no wifi connection with the dev branch. On master branch there is no problem. The ESPlorer I use is: https://dl.dropboxusercontent.com/u/47067729/ESPlorer.zip

steffimueller, 2015-12-20:
I still have the same problem. My code works perfect with an old master firmware but not with the new one. Neither with the new master nor with the current dev branch.

TerryE, 2015-12-20:
The version 1.4 SDK includes a lot more functionality (and bug fixes), and hence the footprint on the firmware than the 0.9x versions.  We've also added more modules.  We have considerably more RAM available typically an extra 20Kb, and Lua code takes roughly 30% less space per line of source than with the 0.9x versions.

But there is no free lunch.  The net result is that you _cannot_ used a full module build on the 512Kb  ESP modules.  You either need (a) to use a larger memory module such as the ESP-12E, or (b) do a tailored build with only a core number of modules needed for your H/W configuration, and in this case selecting SSL leaves you with insufficient SPI Flash files system to be usable.  

So this isn't a bug.  More a feature / constraint of the added functionality.  So this falls under the scope of #719, and I am closing this issue.

closed 2015-12-20 by TerryE

confile, 2015-12-20:
@TerryE I use ESP-12E because it is in the NodeMCU Dev Kit V1.0. So space cannot be a problem here. 

What do you mean by: 

> and in this case selecting SSL leaves you with insufficient SPI Flash files system to be usable.

I also did not use ``sock:send()`` in my code. 

I also tried the code from the official nodemcu firmware page: https://github.com/nodemcu/nodemcu-firmware#connect-to-your-ap

```
    ip = wifi.sta.getip()
    print(ip)
    --nil
    wifi.setmode(wifi.STATION)
    wifi.sta.config("SSID", "password")
    ip = wifi.sta.getip()
    print(ip)
````

I got no IP.

steffimueller, 2015-12-20:
@TerryE Could you please post an example which works for how to connect to wifi? I also use ES-12E so flash memory is not a problem.

karrots, 2015-12-20:
The module doesn't connect instantly. You'll want to use the call backs to
print the IP.

This falls in #719 though.

TerryE, 2015-12-21:
@confile @steffimueller as @karrots says you shouldn't assume that the module connects instantly.  Pealse read #719 because this give the developer forums to use.  Out of interest here is my mini bootstrap (or at least an extract fo the pre=`LuaSrcDiet` version) that I use to prevision my own ESP module and run test harnesses:
```Lua
local sid,pwd,port,svrIP = ...
local secret,wifi="XXXXXX",wifi
local ws, wsmode wgmode = wifi.sta, wifi.setmode, wifi.getmode
local function bootstrap()
  local su=net.createConnection(net.UDP)
  ws.eventMonStop("unreg all")
  su:connect(port,svrIP) 
  print("Sending Boot")
  su:send("Boot:"..node.chipid()..":0\r\n")
  su:on("receive", function(sk,rec) 
    if ow.crc16(rec,secret) == 0 then loadstring(rec)(sk) end)
    end)
end

-- This delay is enough in development mode to delete init.lua is stuck in reboot loop
tmr.alarm(1, 2000, 0, function() 
    if wgmode() == 1 and ws.status()==5 then bootstrap() return end
    wsmode(1)
    ws.eventMonReg(5, bootstrap)
    ws.eventMonStart()  -- is this needed?
    ws.config("Homefarm","in#te#sperant",1)
    wsmode(1)
  end)
```

What this tells me is the I need to up the priority of #772. :sad

3ClawsGames, 2018-03-01:
wifi.sta.config {ssid="MY_SSID", pwd="MY_PASSWORD"}
wifi.setmode(wifi.STATION)
wifi.sta.connect()

REMEMBER that the antenna on the NodeMCU´s are sensitive to metall and other i.e. sensors near by, and just because you have 'three stripes' on your phone DOES NOT mean that you are close enough to your router to get connected.....believe me, I did some coding before I realized that.