nodemcu/nodemcu-firmware #865: Waiting answer from ESP - Timeout reached. Command aborted.Waiting answer from ESP - Timeout reached. Command aborted.

confile, 2015-12-20:
When using Nodemcu 1.4 master and ESPlorer doing "Send to ESP" I often get the following error:  

```
Waiting answer from ESP - Timeout reached. Command aborted.Waiting answer from ESP - Timeout reached. Command aborted.
```

I tested the same with master branch all 31 modues with SSL on and no debugging: nodemcu-master-31-modules-2015-12-20-15-38-04-float.bin.

I created the firmwares with this tool: http://nodemcu-build.com/

Using ESplorer to upload the init.lua script I get no wifi connection with the dev and master branch. The ESPlorer I use is: https://dl.dropboxusercontent.com/u/47067729/ESPlorer.zip

The code I uploaded was: 

```
WIFI_SSID = "ssid"
WIFI_PASS = "123213123123"

ip = wifi.sta.getip()
print(ip)

wifi.setmode(wifi.STATION)
wifi.sta.config(WIFI_SSID, WIFI_PASS)
wifi.sta.connect()
ip = wifi.sta.getip()
print(ip)
 
ledPin = 4  -- blue led
ledRedPin = 0  -- red pin

gpio.mode(ledPin,gpio.OUTPUT)
gpio.mode(ledRedPin,gpio.OUTPUT)

function led(pin, state)
  if (state == 0) then
   gpio.write(pin, gpio.HIGH)
  else 
    gpio.write(pin, gpio.LOW)
  end
end

 while 1 do
  led(ledPin,0)
  tmr.delay(1000000)   -- wait 1,000,000 us = 1 second
  led(ledRedPin,0)
  led(ledPin,1)
  tmr.delay(100000)   -- wait 1,000,000 us = 1 second
  led(ledRedPin,1)
  led(ledPin,0)
  tmr.delay(1000000)
  led(ledRedPin,0)
  led(ledPin,1)
  tmr.delay(1000000) 
   print(ip)

end
```

It only works when I do "Save to ESP". Why does it not work with "Send to ESP"?

ingie, 2015-12-20:
it may not be what's happening - as I can't see why one way would work and not the other - but your while loop is a very cpu hanging block due to the use of tmr.delays - which are blocking calls from my reading of the documentation.

you'd be better off using tmr.alarms to do that "loop" and use a counter in the alarm to increment and print out the ip once the alarm (your 1 LED blink *)  has fired X times.

* (a single led Blink on or off - don't do both and use tmr.delay to pause to make it visible, use negation of the prior state to flip them on/off)

basically: *don't* use tmr.delays if you're actually waiting for something to happen :)

from the doc: "delay us micro seconds. Note: tmr.delay() will make CPU work in monopoly mode, so other instructions would be blocked. Take care in using this function."

as i say, whether this is the problem you're having or not, i don't know... but it might help :)

ingie, 2015-12-20:
in case I didn't explain that well, or in case someone else comes by and doesn't understand what I mean, i'll post an example of that way of doing it here: ( but it's not optimised for storage space, just for readability ) if -to the OP- I explained it fine, but that didn't fix it... I apologise for being either useless, patronising, or both - as I'm not intending such :)

```lua
local MAX_WAIT = 6 -- number of flashes before calling final function
local ALARM_ID = 0 -- an alarm id so we can stop the one we start 
local REPEAT = 1   -- 1 = repeat, we want to repeat
local SLEEP = 1000 -- time between flashes

local ledState = { { pin = 4, state = gpio.LOW } , { pin = 0, state = gpio.HIGH } }

local function flashLEDS() -- swap the state of the two LEDs
    for _,led in pairs(ledState) do
        led.state = led.state == gpio.LOW and gpio.HIGH or gpio.LOW
        gpio.write(led.pin,led.state)
    end
end

local function finalFunction()
    print("It's the final function...")
end

local count = 0
tmr.alarm(ALARM_ID, SLEEP, REPEAT, function() 
    count = count + 1
    flashLEDs()
    if count > MAX_WAIT then
        tmr.stop(ALARM_ID)
        finalFunction()
    end  
end)
```

jmattsson, 2015-12-20:
As @ingie already pointed out, you're running an infinite loop. When esplorer does the Send to ESP, it expects to get a prompt back. However, you just told the Lua interpreter to run forever (actually, until the watchdog forces a hardware reset), so it never returns the prompt, and esplorer complains.
#dontdothat

Closing. Not a bug.

closed 2015-12-20 by jmattsson

TerryE, 2015-12-21:
Please see #719