nodemcu/nodemcu-firmware #868: Calling natives is slow ?

Alkorin, 2015-12-21:
I saw a performance bump with the lasts merges (20ms won on a 37ms loop that's nice !), but it's still too sloow for me :/

I've a simple call which set the color value of a LED in my userdata.
Here are some different way to call my C subs, gpio calls are there for timing with my DSO.

I put in the comment timing with 68eadaaf4596a112144c3b2e1dc52a04c253601b at the left, and timings with current dev branch 1462d00e9f1ee8988a588076bd3eeb38afe7b8e7.

```
-- 98ms => 75ms
local a = ws2812.newBuffer(300); tmr.alarm(0, 500, 1, function()
    gpio.write(3, 1)

    for j=0, 299 do
        ws2812.set(a, j, 0, 0, 0)
    end
    
    gpio.write(3, 0)
end)

-- 53ms => 34ms
local a = ws2812.newBuffer(300); tmr.alarm(0, 500, 1, function()
    gpio.write(3, 1)

    for j=0, 299 do
        a:set(j, 0, 0, 0)
    end
    
    gpio.write(3, 0)
end)

-- 37ms => 18ms
local a = ws2812.newBuffer(300); tmr.alarm(0, 500, 1, function()
    gpio.write(3, 1)

    local g = ws2812.set
    for j = 0, 299 do
        g(a, j, 0, 0, 0)
    end
    
    gpio.write(3, 0)
end)
```

A full native call is at least 70 times faster, but I rarely fill my led stripe with the same color ... ;)

```
-- 260us
local a = ws2812.newBuffer(300); tmr.alarm(0, 500, 1, function()
    gpio.write(3, 1)

    a:fill(0, 0, 0)
    
    gpio.write(3, 0)
end)
```

Any idea / hints to speedup my calls ? Is it possible to dump the generated bytecode to understand how code is called and the difference between the versions ?

Just to see if I continue to use/code the "object oriented" versions which are easier to read, but two times slower :(

TerryE, 2015-12-22:
I've take this as a module developers Q otherwise its into the #719 bit-bucket, I'm sorry to say.  Most of what you are saying standard lua performance stuff.  You need to look at the luac listings etc.  And one day I'll write a tutorial on all this.  The 100× VM instructions to C or thereabouts is pretty standard.  It you want to drop this then you need a smarter API,  for example using  RBG char strings as you only pass 1 char*3 param rather than 3 ints and or introduce a run length scheme, e.g. 
```Lua
a:vectorfill {
-- start, runlength, RBG,
0, 10,  '\0\255\255',     -- 10 cyan lights,
10, 10, '\255\255\0',     -- 10 yellows
-- ...   
}
```
Try developing this and take some benchmarks.

TerryE, 2015-12-22:
@Alkorin, this is too general to stay open.  If you have a specific issue then raise a it as a new issue.

closed 2015-12-22 by TerryE