nodemcu/nodemcu-firmware #870: WiFi event monitor: Upgrade to SDK event monitor

dnc40085, 2015-12-22:
I'm looking to get some input on the API (and corresponding code) I have written to expose the event monitor added with esp_iot_sdk_v1.0.1_15_04_24 before I submit a pull request.

Here's a link to the branch: [dev_new_event_monitor](/dnc40085/nodemcu-firmware/tree/dev_new_event_monitor)

Currently, `wifi.eventmon.start()` needs to be executed once by the user to register the C callback with the SDK. I would like to have it so it automatically registers the CB on boot but I'm not exactly sure how to do that.

# wifi.eventmon sub-module

## CONSTANT
wifi.eventmon.stamode_connected, wifi.eventmon.stamode_disconnected, wifi.eventmon.stamode_authmode_change, wifi.eventmon.stamode_got_ip, wifi.eventmon.stamode_dhcp_timeout, wifi.eventmon.softapmode_staconnected, wifi.eventmon.softapmode_stadisconnected, wifi.eventmon.softapmode_probereqrecved

## DISCONNECT REASON CONSTANT
wifi.eventmon.reason.unspecified, wifi.eventmon.reason.auth_expire, wifi.eventmon.reason.auth_leave, wifi.eventmon.reason.assoc_expire, wifi.eventmon.reason.assoc_toomany, wifi.eventmon.reason.not_authed, wifi.eventmon.reason.not_assoced, wifi.eventmon.reason.assoc_leave, wifi.eventmon.reason.assoc_not_authed, wifi.eventmon.reason.disassoc_pwrcap_bad, wifi.eventmon.reason.disassoc_supchan_bad, wifi.eventmon.reason.ie_invalid, wifi.eventmon.reason.mic_failure, wifi.eventmon.reason.4way_handshake_timeout, wifi.eventmon.reason.group_key_update_timeout, wifi.eventmon.reason.ie_in_4way_differs, wifi.eventmon.reason.group_cipher_invalid, wifi.eventmon.reason.pairwise_cipher_invalid, wifi.eventmon.reason.akmp_invalid, wifi.eventmon.reason.unsupp_rsn_ie_version, wifi.eventmon.reason.invalid_rsn_ie_cap, wifi.eventmon.reason.802_1x_auth_failed, wifi.eventmon.reason.cipher_suite_rejected, wifi.eventmon.reason.beacon_timeout, wifi.eventmon.reason.no_ap_found, wifi.eventmon.reason.auth_fail, wifi.eventmon.reason.assoc_fail, wifi.eventmon.reason.handshake_timeout

## wifi.eventmon.register()
#### Description
Register/unregister callbacks for WiFi event monitor

#### Syntax
wifi.eventmon.register(Event, function(T))
wifi.eventmon.register(Event, "unreg")

#### Parameters
Event: WiFi event you would like to set a callback for 
* Valid WiFi events:
  * wifi.eventmon.stamode_connected
  * wifi.eventmon.stamode_disconnected
  * wifi.eventmon.stamode_authmode_change
  * wifi.eventmon.stamode_got_ip
  * wifi.eventmon.stamode_dhcp_timeout
  * wifi.eventmon.softapmode_staconnected
  * wifi.eventmon.softapmode_stadisconnected
  * wifi.eventmon.softapmode_probereqrecved

#### Returns
nil

#### Example

```lua
 wifi.eventmon.register(wifi.eventmon.stamode_connected, function(T) 
 print("\n\tSTA - CONNECTED".."\n\tSSID: "..T.SSID.."\n\tBSSID: "..
 T.BSSID.."\n\tChannel: "..T.channel)
 end)
 
 wifi.eventmon.register(wifi.eventmon.stamode_disconnected, function(T) 
 print("\n\tSTA - DISCONNECTED".."\n\tSSID: "..T.SSID.."\n\tBSSID: "..
 T.BSSID.."\n\treason: "..T.reason)
 end)

 wifi.eventmon.register(wifi.eventmon.stamode_authmode_change, function(T) 
 print("\n\tSTA - AUTHMODE CHANGE".."\n\told_auth_mode: "..
 T.old_auth_mode.."\n\tnew_auth_mode: "..T.new_auth_mode) 
 end)

 wifi.eventmon.register(wifi.eventmon.stamode_got_ip, function(T) 
 print("\n\tSTA - GOT IP".."\n\tIP: "..T.IP.."\n\tMASK: "..
 T.MASK.."\n\tGW: "..T.GW)
 end)

 wifi.eventmon.register(wifi.eventmon.stamode_dhcp_timeout, function(T) 
 print("\n\tSTA - DHCP TIMEOUT")
 end)

 wifi.eventmon.register(wifi.eventmon.apmode_staconnected, function(T) 
 print("\n\tAP - STATION CONNECTED".."\n\tMAC: "..T.MAC.."\n\tAID: "..T.AID)
 end)

wifi.eventmon.register(wifi.eventmon.apmode_stadisconnected, function(T) 
 print("\n\tAP - STATION DISCONNECTED".."\n\tMAC: "..T.MAC.."\n\tAID: "..T.AID)
 end)

wifi.eventmon.register(wifi.eventmon.apmode_probereqrecved, function(T) 
 print("\n\tAP - STATION DISCONNECTED".."\n\tMAC: ".. T.MAC.."\n\tRSSI: "..T.RSSI)
 end)

 wifi.eventmon.start()
```

TerryE, 2015-12-22:
Firstly use `lua_getstate()`.

Next the convention is that constants are uppercase. and I think that the 'mode' is superfluous, so why not 
`wifi.eventmon.STA_GOT_IP` etc.?  Then this will be consistent  with other modules.`

We also need to document T.  Sorry but I am a bit snowed under at the moment so I can't do a decent code review, but if we do spin off a temporary `dev150` then why son't we consider merging in this PR to this branch to make it easier to kick the tyres".

Third, I feel that `ẁifi.c` is getting a little bloated.  Maybe it's spin this off into a wifi folder. and even make it an optional module.

jmattsson, 2015-12-23:
Over all I'm in favour here! Terry already made some excellent suggestions, so I'll just add another couple here:
- I'd like to see either an `unregister` function, or just have the `register` function take a `nil` for the function argument to deregister. Passing a string is very non-standard.
- You should be able to register the event callback inside the Lua module init function (look for `luaopen_xyz` in the existing modules for examples).

And finally a question: Will this replace the earlier eventMon{Reg,Start,Stop} functions?

nickandrew, 2015-12-23:
+1 Looks good.

dnc40085, 2015-12-25:
Thank you for your input! 
@TerryE 
* 1.Thanks for the confirmation, I added `gL=lua_getstate()` to `luaopen_wifi` and removed all instances of `gL=L` in the wifi module so they wouldn't interfere.
* 2a. Done. Should I un-abbreviate the constants or should I leave them be?
* 2b. Done.
* 3. I've been thinking that for a while. 
I tried to do this but I can't figure it out, I put the code in a separate source file and tried using `NODEMCU_MODULE(WIFI_EVENTMON, "wifi.eventmon", wifi_eventmon_map, luaopen_wifi_eventmon);` and it told me `wifi.eventmon` didn't exist. so then I tried putting the define for `wifi_eventmon_map` in a header file and inlcuded it in both `wifi.c` and `wifi_eventmon.c` and ended up getting a linker error saying `wifi_eventmon_map` not defined. 

Also, I couldn't figure out how to make files compile when I put them in the new folder `app/modules/wifi`. I figure that the new folder needs a makefile, but I'm not skilled in the art of makefile-fu so I don't know.

@jmattsson 
* Done. I can't figure out why I did that in the first place. 
Would it be preferred to have a `wifi.eventmon.unregister(event)` function instead of `wifi.eventmon.register(event, nil)`?
* Thank you for the info, I always wondered what `luaopen_xyz` was for.
Just out of curiosity, is `luaopen_xyz` executed on boot or when the module is first accessed?
* It was my intention to replace `wifi.sta.eventmon` with the new one... But if it would be preferred to keep it and have them both coexist, I don't see a problem with that.
I just didn't like that the old one polled `wifi_station_get_connect_status()` constantly, then triggered the user callback if it changed, it unnecessarily loads the cpu down limiting how fast the user's application can run.

TerryE, 2015-12-25:
Merry Christmas!! I'll reply tomorrow  :smile:

dnc40085, 2015-12-25:
Thank you, Merry Christmas to you as well.

TerryE, 2015-12-26:
A quick comment on using the `lua_openXYZ()` hook: you need to be a little careful here if your module depends on others having done their initialisation, because there is no guarantee as to the order in which these are called, so some modules may need to defer some initialisation processing to a timer or task callback, but this isn't the case here, I think.

dnc40085, 2015-12-28:
I figured out how to split the event monitor off into a separate file, not sure if it is the right way, but it works!

Now to figure out how to get the files to compile in a subdirectory of modules...

TerryE, 2015-12-28:
This is standard stuff -- recursion magic in make. I can do this for you.

dnc40085, 2016-05-02:
I forgot to close this.

closed 2016-05-02 by dnc40085