nodemcu/nodemcu-firmware #872: Changing the default user settings.

TerryE, 2015-12-23:
At the moment the "default" `dev` compile doesn't load on the vast majority of ESP modules out there (the 512Kb parts).  My recommendation is that we change the default in user_config.h and user_modules.h so that if  a developer downloads the repo dev branch or zip and does a make then the binary is usable on a 512Kb part.  

Anyone who needs to add anything can uncomment the relevant line and do the make again.  My suggestion for defaults (in my `cnf` format are):
```
bit_rate               = 115200
client_ssl             = disable
gpio_interrupt         = enable
spiffs_cache           = enable
optimize_debug         = 2
led_high_count_default = 10
led_low_count_default  = 0

builtin = string,table,coroutine,math,debug_minimal
modules = node,file,gpio,wifi,net,i2c,spi,tmr,adc,uart,ow,bit
```
but in terms of `//` edits of the `user_*.h` files of course.

jmattsson, 2015-12-23:
I guess removing SSL saves a lot of space? It feels a bit wrong to remove it, but I can see an argument for it.

Module list looks okay. If anything, I'd suggest also including rtctime since the default timekeeping on the ESP is rather horrible.

nickandrew, 2015-12-23:
As I have never used `i2c,spi,adc,bit` modules I'd suggest removing them from the default config :smile:

I also suggest removing `ow` as IMHO the point of a default config is like a demo mode; it should be enough to boot up the NodeMCU and write some Lua scripts and do some networking; onewire isn't crucial for that.

TerryE, 2015-12-23:
I'd keep bit, and i2c.  but that's just my one vote.

devsaurus, 2015-12-23:
> bit_rate = 115200

Nice - the change should be highlighted in the Readme. 

+1 for bit - it's a valuable substitution for the missing & and | operators.

i2c, spi, ow, adc, and gpio are general purpose interface modules. If we can't set a prio then either keep or remove them all. I'm for keeping them as @TerryE suggested.

TerryE, 2015-12-23:
I feel that 64Kb is about the smallest usable SPIFFS, and there is no way to achieve this it if we include SSL or SHA2.

pjsg, 2015-12-23:
Can we have a define to indicate that we are targeting the large flash parts which includes much more stuff. I.e. if I add #define FLASH_LARGE (or similar), then I get an expanded set of modules -- maybe all of them.

marcelstoer, 2015-12-23:
I fully agree with Terry's initial list except for the modules. 

IMO there shouldn't be any guessing as for SSL and the modules. Why don't we strictly go by http://nodemcu-build.com/stats.php? Dumping SSL should be a no brainer as 70% of the general NodeMCU user crowd doesn't use it. I'd cut the modules list either right after 1-wire or after bit.

When we change the default bit rate we should also do away with the dedicated developer bit rate.

karrots, 2015-12-23:
My guess is 90% don't use SSL because it's a bit of black magic to get it
to work correctly. Not really saying it shouldn't be dumped. The network &
sysadmin side of me says don't dump it nothing should be in the clear but I
realize with this device that isn't a reality due to the SDK when it comes
to SSL.

TerryE, 2015-12-24:
> Can we have a define to indicate that we are targeting the large flash parts which includes much more stuff. I.e. if I add #define FLASH_LARGE (or similar), then I get an expanded set of modules -- maybe all of them.

Unfortunately -- in this version at least -- we are talking about adding `//` or not in front of a selected number of modules.  If we start allowing configuration groups the complexity will reun away with us.  My view is that the default build is to get the new-to-ESP to begin to develop code.  Much past that they should use a custom build. 

> When we change the default bit rate we should also do away with the dedicated developer bit rate.

+1 on that its a complexity added because the default rate is 9600.  Without it, naaa

marcelstoer, 2016-03-01:
I started the implementation in my own branch and commit these changes, see 193237fb13773d04ee873ea1bae23e1023e51b6b. I tested the firmware - so far so good.

> i2c, spi, ow, adc, and gpio are general purpose interface modules.

marcelstoer, 2016-03-06:
@ fellow collaborators, do you want to continue finding a consensus on the most sensible modules list here or shall I raise https://github.com/nodemcu/nodemcu-firmware/commit/193237fb13773d04ee873ea1bae23e1023e51b6b as a PR and we'll do it there?

devsaurus, 2016-03-06:
Thanks for bumping this topic Marcel.

I'd be fine with adding MQTT and DHT to the default list. A careful selection would have been important for binary releases, but since these went out of fashion there's not a huge value in this list anymore IMO.
Increasing the default baudrate for sure is a noticeable change...

TerryE, 2016-03-07:
I agree with Nick's point about ow being a nasty implementation but to me this means that we should fix ow soon rather than omit it.  DHT is another nasty bit-banging implementation that will kill the WiFi stack, so again we need to sort this, but that's a separate argument from what should be default.  Marcel's 193237f is a step improvement over the current defaults, IMO, so if we want to make this change sooner rather later then let's just raise the PR, close this issue and move any remaining discussion to it.

PR #1127 (2016-03-09): Change a number of default values
fixes #872
closed 2016-03-09 by marcelstoer