nodemcu/nodemcu-firmware #873: A new year wish list for core firmware activities

TerryE, 2015-12-23:
II just wanted to float this as a general discussion for committers and our regular contributors to brainstorm some ideas for activities that we might consider for a first quarter improvement programme.   I have three that I haven't got open issues for but would be interest in feedback on.

*    **Dumping dead code**.  When zeroday did his first port of eLua, varoius chunks of code were `#ìf 0` commented out and also blocks of `//`.  OK some light the debug and math functionality might be reintroduced at a later stage so maybe this merits being left, but a lot is literally fossil code.  My view is that we should have a policy of getting rid of this from tghe current working source, and use git diff etc for looking at code archaeology.

*  **simplify unnecessary prefixes**. We've the `c_` prefixes for the string routines, We've `malloc`, `c_malloc, `os_malloc`, `mem_alloc`, etc. these are mostly synonyms  for their equivalents.  I would prefer the firmweware just to stick to the standard `strlen`, `malloc` etc.  In terms of mallocing we should only have a a separate name if we want different functionality for example `luaM_malloc(L,t)` does a malloc but throws a Lua error if the malloc fails.  It would really make working with the firmware base a lot simpler if we didn't have all of these redundant synonyms around the place.

*   **Tidy up the IRS code in `app/driver`**.  some of these modules has functions that have nothing to do with ISRs or realtime functionality (e.g. the ow crc routines)  IMO, this modules should the true core time-critical stuff in them -- absolutely the minimum needed here,  and in this case there is a good argument for moving them into `iram1` to guarantee that we aren't getting FLash cache miss and exception handler overheads impacting timig.

Any other suggestions / wants and any volunteers willing to step up and do the work?

dnc40085, 2015-12-26:
Shouldn't we use ROM functions where possible to avoid unnecessarily using flash space?

TerryE, 2015-12-26:
Yes, but there are always valid exceptions which is why iram1 is there in the first place.  An ISR is supposed to complete it under 50µS, so we don't want to get exception handlers to fill the cache, or handle unaligned access getting triggered during an interrupt.  They will fail anyway if interrupts are masked.

So IMO, any routines which have code regions protected by `noInterrupts()` -- `interrupts()` guards should run out of iram1.  We should keep this code absolutely to the smallest possible, so this shouldn't be a sizing issue.

dnc40085, 2015-12-26:
I was referring to your second bullet point. Shouldn't we try to use functions listed in `eagle.rom.addr.v6.ld` (mem_malloc, mem_free, etc.)?

TerryE, 2015-12-26:
We do. The various forms are synonyms for the ROM routines in the main. But eLua a S Lua us the standard names SGI why should or version use non-standard ones?

TerryE, 2015-12-26:
We do. The various forms are synonyms for the ROM routines in the main. But eLua and Lua use the standard names so why should we use non-standard ones?

dnc40085, 2015-12-26:
Oh ok I see, That makes sense.

vowstar, 2016-01-01:
https://github.com/pvvx/MinEspSDKLib

May be we can remove the libmain.a and libc, replace to open-source one.

TerryE, 2016-03-17:
"Simplify unnecessary prefixes" now covered by #1165.

TerryE, 2016-04-11:
We need to move the "dumping dead code point into our module guideline.  The other two points have been done, so let's close this.

closed 2016-04-12 by marcelstoer