nodemcu/nodemcu-firmware #875: DHCP Device name contains illegal underscore in development build

boatmon0, 2015-12-23:
In the development build the default device name my dhcp router shows for me is ESP_058CDC.  The underscore in the name seems to mess up the router and the device is not reachable by that name. (ping or http://)  I built 12/22/2015 at nodemcu-build.com, installing their default packages.  
Please replace the underscore with a dash.

Is it possible for me to change the device name?  I would like to provide my own name. This device name should probably default to the device id, to maintain uniqueness.

[image]

devsaurus, 2015-12-23:
The default dhcp hostname is constructed automatically without means to influence the composition. It could be set to an arbitrary string via [wifi_station_set_hostname()](https://github.com/nodemcu/nodemcu-firmware/blob/master/sdk/esp_iot_sdk_v1.4.0/include/user_interface.h#L252) but there's no Lua binding for this function yet. This might involve some trial and error to get it working.
Since we're currently focusing on basic topics in the firmware, there's probably no activity at the moment to integrate new SDK functions. A pull request with such an enhancement is always welcome, though.

udif, 2015-12-23:
http://stackoverflow.com/questions/4919889/is-there-a-standard-that-defines-what-is-a-valid-ssid-and-password

It seems that there is no such thing as an "illegal" character in an SSID. As such, the nodemcu behavior is completely legal. you may want to override the default name, but it is definitely not "illegal".

TerryE, 2015-12-23:
As @devsaurus says, this is SDK functionality so you really need to report the bug to Espressif.  It's an enhancement to add an extra API to override the default.

Where in RFC2131 does it forbid the use of _ ?

udif, 2015-12-23:
sorry, I misread your issue, because the SSID is also of the same form, ESP_xxyyzz, where xxyyzz are the 3 lower bytes of the MAC address.

You are talking ABOUT the DHCP name, not SSID.

boatmon0, 2015-12-23:
Wow, what an active group.  Great stuff, and you have at least pointed me in the right direction.  

I found the problem when I named a device using the underscore.  Here is my reference.

https://en.wikipedia.org/wiki/Hostname

The Internet standards (Requests for Comments) for protocols mandate that component hostname labels may contain only the ASCII letters 'a' through 'z' (in a case-insensitive manner), the digits '0' through '9', and the hyphen ('-'). The original specification of hostnames in RFC 952, mandated that labels could not start with a digit or with a hyphen, and must not end with a hyphen. However, a subsequent specification (RFC 1123) permitted hostname labels to start with digits. No other symbols, punctuation characters, or white space are permitted.

Perhaps I misunderstand, but my tplink router will not resolve local domains if the esp8266 is turned on. 

Thanks Again

Eugene

karrots, 2015-12-23:
As the wikipedia article mentions underscores are not a supported character not only in hostnames but in  DNS in general. Hence the reason the TP-link will not resolve it.

devsaurus, 2015-12-24:
> Perhaps I misunderstand, but my tplink router will not resolve local domains if the esp8266 is turned on.

Wow, sounds like a major troublemaker for such a network configuration. @boatmon0 consider a bug report at [Espressif](http://bbs.espressif.com/viewforum.php?f=66), they're to change the default naming.

Should we raise the prio for this "enhancement" then? Just to provide a workaround for a known limitation which we inherited from the SDK. I could dig into this during the next days.

boatmon0, 2015-12-24:
Well, @devsaurus  I can go back to the production version and all is good.  I foolishly thought that this was a one line change and dived into the compile game and am now stuck.  Perhaps someone can point me in the proper direction.  I have a debian7 VM that I use for some other development work.  So I used https://github.com/pfalcon/esp-open-sdk to get the toolchain.  Was that the best way?

Next I downloaded a zip of nodemcu-firmware development branch.  Commented out a couple of things in user_modules.h, and did the make. In my bin directory I have a file 0x10000.bin that is 464.1kB, and another file 0x0000.bin which I ignored.  When I shoot that 0x10000 file down to the esp8266 and reboot, I get something that looks like garbled error messages.  Did I miss a step?

Meanwhile, thanks fore the expressive link, and I'll head over there and see if they can help.

Best Regards

TerryE, 2015-12-24:
Have you thogh of using a static IP as a workaround for mow?  Most router hubs will handle a classC address but only use half or so for a DHCP pool.  On my router for example addresses up to .64 can be used for static allocation.

boatmon0, 2015-12-24:
Yes, that is the easy way to solve the problem, perhaps that is best at this point.

marcelstoer, 2015-12-25:
> ...dived into the compile game and am now stuck

Take a look at https://github.com/nodemcu/nodemcu-firmware#building-the-firmware

TerryE, 2015-12-25:
If you only have 512Kb modules then you definitely need to drop SSL in your config, and remove any modules that you aren't using and the command is (from the root nodemcu directory):
```bash
pathToTool/esptool.py -p /dev/ttyUS*  write_flash 0x00000 bin//0x00000.bin  0x10000 bin/0x10000.bin
```
So you definitely do _not_ need to ignore the 0x0000.bin file, but this is the point I refer you to #719.  You really need to get this general advice n other forums.

marcelstoer, 2016-01-17:
> I am keeping this issue open solely to track adding a Lua binding for `wifi_station_set_hostname()`. 

Was introduced with #878. Closing this one.

closed 2016-01-17 by marcelstoer

boatmon0, 2016-01-19:
Thanks for the fix!  It works great!