nodemcu/nodemcu-firmware #876: sntp module

iMihael, 2015-12-24:
Hello, I saw in documentation sntp module and tried to use it:
`sntp.sync('88.147.254.232', 
    function(sec,usec,server)
        print('sync', sec, usec, server)
    end
    )
`
but failed with error:
`attempt to index global 'sntp' (a nil value)`

I'm trying to search for sntp module in source code but I did not find any mention of it.
I'm using *0.9.6-dev_20150704* release of nodemcu firmware, builded from sources.

nickandrew, 2015-12-24:
Please see #719.

TerryE, 2015-12-24:
As Nick implies you need to be using a current build. The references give you the links.

closed 2015-12-24 by TerryE

fulvous, 2017-02-20:
Hi bolsunovskyi you just need to rebuild your firmware enabling the sntp module. It seems it can be done on the cloud service, otherwise, if you are building by hand follow the next steps:

1) go to your firmware folder
2) edit the following file: app/include/user_modules.h
3) uncomment the following line:
#define LUA_USE_MODULES_SNTP
4) recompile with make
5) flash the new firmware.

I hope this helps