nodemcu/nodemcu-firmware #877: ws2812 stops working after a few writes

sjoerdtimmer, 2015-12-25:
I'm having 'curious' issues with my 72 led ws2812 ledstrip.
It works at first, but after a few writes it stops working: leds go all off and nothing can turn them back on. Not even rebooting or even reflashing nodemcu firmware. So far I have found one way to make the strip work again:
1. flash some other custom c-firmware that addresses the leds (this is an old version of the project and I wished to add a simple webserver to it which I thought was gonna be easy using nodemcu;-))
2. flash nodemcu again, upload lua script and run
this will work for a couple of calls to ws2812.writergb() after which the system breaks again.

This suggests that either the ws2812 is put in a non-responsive state or some pretty low-level register-stuff gets mixed up...

Any suggestions what might be causing this?
Any debugging hints?

sjoerdtimmer, 2015-12-25:
BTW:
I tried with different firmware build from nodemcu-build.com. The one I have been testing with lately is:

    NodeMCU custom build by frightanic.com
    branch: dev
    commit: 0069f002a8686a7561ea40234327ab53b42f00d4
    SSL: false
    modules: node,file,gpio,wifi,net,tmr,uart,ws2812,cjson,rtctime,sntp,enduser_setup

This is my lua script, if you're interested:

    function showall(r,g,b) 
        NUMLEDS = 72
        red = {}
        green = {}
        blue = {}
    
        for i=1,NUMLEDS do
            red[i] = r; 
            green[i] = g; 
            blue[i] = b;
        end
    
        bytes = {}
        for i=1,NUMLEDS do
            bytes[3*i-2] = math.floor(red[i])
            bytes[3*i-1] = math.floor(green[i])
            bytes[3*i  ] = math.floor(blue[i])
        end
    
    
        rgb = string.char(unpack(bytes))
        ws2812.writergb(3, rgb)
    end

marcelstoer, 2015-12-25:
Are you suggesting there's a bug in NodeMCU ws2812? If not, then please take note of #719.

TerryE, 2015-12-25:
You are using large (e.g. up to 216 elt) temporary arrays so will have garbage collection issues, but advice on Lua techniques is this is outside the scope of this forum (See #719).  My suggestion is that @Alkorin @bbktsk (who has another open PR #869 open) have a look and comment re the driver aspects.

kbeckmann, 2015-12-26:
@sjoerdtimmer First, I'd recommend you to use ws2812.write() because it's faster and uses less memory. The actual bitstream that's sent to the ws2812 has to be in GRB order and this call will just send the raw data immediately. It will be faster and save memory.

I tested your code and it does use some memory but it doesn't leak any (checked with node.heap()), so I don't think that's the problem. Make sure that you have enough free heap to begin with since the writergb() function allocates (and frees) a copy of the input string, in order to flip R and G.

I have to be honest, I haven't tested the writergb() code extensively, but it did work pretty well when I tried it out.

I am personally using ws2812.write() to power my 8*144 led banner (1152 pixels) in over 20 fps and it looks great and doesn't crash (tried it today with latest dev branch). So I'd recommend you to try with that function first. If the problem is still there, then the problem is not in the ws2812 code.

Also power cycle your led strip when it acts up, it might help.

sjoerdtimmer, 2015-12-26:
Thanks for all the good suggestions!

I am indeed suggesting this may be a bug in nodemcu firmware. 

Please correct me if I'm wrong, but to my understanding heap issues should be fixed after node.restart(). Even if (due to out of heap issues) the flash gets corrupted, a clean reflash with nodemcu firmware should recover this issue. 

However, in my case it did not. Only writing to the ws2812 with some native C app for the esp fixed it. This suggests that some persistent config(possibly some gpio register) in the esp8266 gets messed up, which prevents it from updating the ledstrip but has no other symptoms... and which is fixed by my native-c firmware.

I tried with ws2812.write() and I can reproduce the issue. 

I have certainly powersycled the ledstrip many times (leaving it off for as much as a minute) so I think it is safe to exclude the possibility that the ws2812 driver gets into a corrupted state (as that should be fixed after a powercycle). 

I admit that the issue is vague and may or may not turn out to be a bug later.
Any other suggestions about what may be going wrong? how to diagnose/debug?

Alkorin, 2015-12-26:
Is it fixed by a powercycle of the ESP8266 ?

sjoerdtimmer, 2015-12-26:
No, so far I have only been able to fix it by reflashing other firmware, then nodemcu again. This includes at least 2 powercycles. Simply restarting the esp does not fix the issue...

sjoerdtimmer, 2015-12-26:
It seems the problem may be electrical after all...

I just bypassed my logic level shifter and the 470R series protection resistor and all issues are gone now. I have not yet determined whether it is the resistor or the logic level shifter that is messing the signal but it definitely works when I connect the ws2812 directly to gpio0. I guess it's the resistor because it stops working when i bypass the level shifter but put a new series resistor in the line...

What value protection series resistor are you using? I know my ledstrips are cheap chinese ones but this problem is not just with one ledstrip but also with another and even with two entirely different ws2812 modules...

I figured this out because I noticed that the ws2812 also stopped updating temporarily when I connect my logic analyser. This suggests that the signal's impedance is too high? (the logic analyser sees is correctly though) 

If anyone can explain why my custom native-C firmware is not bothered by the series resistance and the lua firmware is, please elaborate... And why the lua firmware works for a few writes and then stops working?? A few mysteries remain...

TerryE, 2015-12-26:
@sjoerdtimmer @Alkorin: Sjoerdt, Thomas, maybe it would be better that you did this 1-1 or on a different forum set up for this type of discussion?  Here isn't the place to discuss electrical design or Lua programming techniques, so I am referencing #719 and closing this as a firmware issue.  

Thomas if you do think that there are material points relating to the firmware itself that can't be addressed on one of the other open ws2812 issues then by all means reopen :smiley: and merry Christmas.

closed 2015-12-26 by TerryE

sjoerdtimmer, 2015-12-26:
Merry christmas to you too Terry!!

The "strange" behaviour seems to be located at the ws2812 side, so I'll close this now (update: thanks Terry for closing this before I could)

writing to it once with no series resistor restores responsiveness... (as does writing to it from my native C firmware). If anyone can explain why I'd be delighted to hear...