nodemcu/nodemcu-firmware #892: Cannot sent 0x00 to UART

Derek-K, 2015-12-30:
I was trying to use the UART communicate to another device with RAW BINARY data. And notice packet truncated when it hits 0x00.

If I am not mistaken, because of the C treat 0x00 as NUL (string terminator) in the source code (uart.c)

May I suggest adding something like UART.BWRITE (id, buffer, len) for sending binary data?

devsaurus, 2015-12-30:
For the moment you could work around this limitation by using integer bytes instead of strings. You can also mix string and integer parameters:
```lua
uart.write(0, 0x11, 0x22, 0x33, 0x44)
uart.write(0, first_string_without_00, 0x00, next_string_without_00, 0x00, 0xaa)
```
Haven't tried it myself, but the source code suggests these (undocumented) integer byte capabilities.

Apart from this, feel free to propose such API extension via a PR.

TerryE, 2015-12-31:
Sounds like a bug to me.  Lua can handle strings with embedded zeros, so its libraries shouldn't use C-style zero byte termination.  I'll add it to my long list.

devsaurus, 2015-12-31:
> Sounds like a bug to me.

I'm not sure - sending embedded \0s in strings works fine all over the place afaik - e.g. in the ws2812 and spi modules.
This small testcase works as expected and feeds all 16 bytes to the terminal:
```lua
s = string.char(0,1,2,3,4,5,6,7,0,1,2,3,4,5,6,7)
uart.write(0, s)
```
@Derek-K please provide more details about how you trigger the truncation with `uart.write()`.
* Firmware version
* Compile settings - either banner string from nodemcu-build service or `user_config.h` and `user_modules.h` if you complied yourself.
* Lua code to reproduce

TerryE, 2015-12-31:
@devsaurus Arnim, it sounds as if you are on top of this, so I'll leave you to take the lead.

devsaurus, 2015-12-31:
Yes, thanks Terry. I should have been more specific in my first post but was too lazy last night :smile:

Derek-K, 2015-12-31:
Sorry, my bad. Long story short, I blame "LuaLoader 0.87" :flushed: 

My initial test using LuaLoader shows
```
> uart.write(0,"Hello\0World!");
Hello
```

But upon testing with RealTerm 2.0.0.70, above code does print a 'nul' in between.

(and I must of reading uart.c when I was half asleep and I can't even find the spot where I think it is C's fault...)

For the record, I just tested using the latest dev build this morning
```
NodeMCU custom build by frightanic.com
	branch: dev
	commit: a8359f1da52993e2dcfd2543cc568b50fda188d5
	SSL: false
	modules: node,file,gpio,wifi,net,tmr,adc,uart,ow,sntp,enduser_setup
 build 	built on: 2015-12-31 15:10
 powered by Lua 5.1.4 on SDK 1.4.0
```

It's all good now, and everyone have a Happy New Year!! :tada:

closed 2015-12-31 by Derek-K

devsaurus, 2016-01-01:
Excellent - Happy New Year as well!