nodemcu/nodemcu-firmware #897: GPIO's not equaly initialised

TheoHarbers, 2016-01-02:
I have an ESP03 board with 5 relay connected to GPIO 12 through 16. Immediately after power on GPIO 12-14 outputs are low (0 = relay off) and GPIO 15-16 are high (1 = relay on). I do not know if there is a reason for this but maybe they are just not initialised the same. As a work around in the INIT.LUA i define them as output and set them to zero and this seems to be fast enough before the 2 relays are activated.

nickandrew, 2016-01-04:
See #719.

Espressif documentation on ESP8266 should explain the GPIO states.

TerryE, 2016-01-04:
@TheoHarbers, just to expand on Nick's point, the Lua firmware makes no special default treatment of the different  GPIOs.  AFAIK, this is entirely down to the hardware specs and the Espressif SDK.  We don't see the "job" of the firmware as to iron out "features" in the underlying H/W and the manufacturer's SDK.

closed 2016-01-04 by TerryE

TheoHarbers, 2016-01-04:
Ok clear and as i wrote before setting them in the INIT.LUA seems to work quite well for slow relays. Although i did not check with a scope to see if there aren't any glitches between power up and execution of the INIT code

dvv, 2016-01-12:
I would vote pro setting all pins to Z or pulled-up inputs at early startup. It is more natural and _much safer_.
(My case, pin 4 which controls a relay, is dont-know-why set high after device reset, and we know nodemcu still suffers sudden resets).