nodemcu/nodemcu-firmware #901: Crash in lmacProcessAckTimeout

pjsg, 2016-01-04:
This is just a placeholder ticket -- the bug is not in the nodemcu code, but (I think) in the espressif SDK. I have opened a bug with them, although the bug is not very reproducible.

There are other people, with non-nodemcu codebases, and with older versions of the SDK that have seen the same crash -- lmacProcessAckTimeout+39 (decimal).

It causes my app to crash roughly once every 4 to 8 hours. It appears to be position dependent -- if I move the board from one location to another, the failure rate goes up or down. The AP is positioned a long way away, so the signal is weak to begin with.

If you have unexplained crashes, then use the improved node.bootreason() from issue #900 to get the failing PC. Then you can look in your mapfile to find the failing function. 

Just to repeat, this is *probably* not a bug in the nodemcu code, but it is a bug that affects the users of nodemcu.

pjsg, 2016-01-04:
Argh! They rejected the bug report with

```
But please report your issue to nodemcu instead, here we accept the bug report based on our original SDK. 
You can report your issue to nodemcu, if they find that is not due to their codes, they will come to us and help us duplicate the issue.
```

nickandrew, 2016-01-05:
@pjsg "they will come to us" :smile: I think "they" (you) already went to "us" (Espressif). Perhaps Terry needs to give the bug report his stamp of approval.

pjsg, 2016-01-05:
I really want to be able to reproduce it easily. The same bug happens with the Arduino / ESP8266 code. The person there has now reported the bug: https://github.com/esp8266/Arduino/issues/1329

pjsg, 2016-01-12:
I just got email from espressif:

Hi,

Please have a try with our latest ESP8266_NONOS_SDK_V1.5.1 http://bbs.espressif.com/viewtopic.php?f=46&p=5315 .

If your problem is still unsolved, please feel free to let us know.

I'm going to try and build with this version and see if the problem has gone away......

pjsg, 2016-01-13:
Yes, the problem seems to have been resolved. My code has been running for 12 hours with no crash.

pjsg, 2016-03-05:
The upgrade solved it.

closed 2016-03-05 by pjsg