nodemcu/nodemcu-firmware #903: How safe is masking all interrupts.

TerryE, 2016-01-05:
We have our bit banging code splattered with `ets_intr_lock()`, `ets_intr_unlock()` sequences, where

  *  the locked code can run to 100s uSec or mSec in some cases.
  *  the code is in Flash so we could get a cache miss interrupt during a locked code sequence.
  *  the code might trigger an unaligned data fetch and the HW will need to dispatch an unaligned data fetch exception to execute the code.

Does anyone know where I can get a detailed description of the rules here? Should we be moving these routines into irom0? @jmattsson, @nodemcu, anyone any comments?

jmattsson, 2016-01-05:
I believe the document you're looking for is the [Xtensa Instruction Set Architecture](http://0x04.net/~mwk/doc/xtensa.pdf).

Based on the testing I did it is possible to hit the unaligned-data-access exception inside an ets_intr_lock() block and have it do its thing fine. Of course, there is a timing penalty, as is there for running code from the flash.

Moving things into iram0 is problematic, as the SDK has gobbled up most of it by now. When I did the 1.4.0 upgrade I used a good portion of my bag'o'tricks to get the NodeMCU bits to fit alongside the SDK bits.

TerryE, 2016-01-05:
The amount of code that truly _needs_ to do interrupt locking is a few 100 bytes. In a few cases we have locking just for the hell of it, such as in the GPIO module where where we locks around a call to a ROM routine which when you look at the code already has locks around the critical region. IMO, any PRs which include this type of code should be rejected unless the author includes a proper impact assessment and justification -- that is if we care about getting the firmware stable.

jmattsson, 2016-01-05:
There should be very few legitimate causes for disabling interrupts. It sounds like it's largely bitbanging which is doing this, so could we get away with a common big-banging routine that takes a list of ops in RAM to bang out, and then have that common routine in iram? I'm thinking along the lines of the Linux SPI driver interface.

TerryE, 2016-01-05:
Now that sounds like an approach that I'd definitely +1. A single unified approach within the core for a common mode issue that a number of libraries face.

nodemcu, 2016-01-05:
+1 too, a function/api that bit bang out with a limited delay(baudrate), in iram.
a little bit late, happy new year!

TerryE, 2016-01-12:
@jmattsson @nodemcu, and anyone else that would like to contribute here.   I've been going through the realtime bits of our implementation and this has thrown up quite a few issues that I think, so rather than starting to implement any changes, I'd prefer to write up some of these conclusions / recommendations for discussion.  Would people prefer that I do this in issue form (e.g. here) or as a document (e.g. on the wiki?)

jmattsson, 2016-01-12:
I'd prefer it in this thread, both for context and easier discussion. Thanks for taking the time Terry!

TerryE, 2016-01-12:
OK, Let's do the points:

1.    The SDK stipulates (Sec 2): _do not call any function defined with ICACHE_FLASH_ATTR in the
interrupt handler_, and in general the supplied ISRs in the `SDK/examples/driver_lib/driver` follow this convention by omitting the `ICACHE_FLASH_ATTR` attribute from ISRs (and the current UART example explicitly states in a comment "**IN NON-OS VERSION SDK, DO NOT USE "ICACHE_FLASH_ATTR" FUNCTIONS IN THE WHOLE HANDLER PROCESS**"), so that the ISRs get correctly loaded into `iram1` By making all text segments default to `irom0`, we've broken this rule.

2. The SDK stipulates (Sec 2): _If interrupts are disabled, then this should be for <10 µSec_ (paraphrasing into English), yet we break this rule regularly in our libraries and terribly in the case of both the GPIO and PWM modules.

3. We sometimes disable all interrupts, when we want to guard against interrupt routine reentrancy and therefore disabling the specific routine should be used.

4. We call Lua callbacks from GPIO ISRs which just breaks every rule in the book.  At a minimum we should be using task breaks

5. Many of our bit banging libraries break both the maximum disabled interrupts period (10µSec) and the general guideline on using software timer callback to do computationally intensive code ("please do not occupy CPU more than 15 mSec").

My recommendation are that: 

*    (A) the `driver` directory is reserved for realtime and h/w interrupt related code -- viz any code which disables any interrupts or ISRs (or called by the same) are placed in driver modules, and v.v. any other routines are removed into the standard `modules` directory (or library folder) to keep driver code to the minimum. This minimal set of code goes into `iram1`

*    (B) wherever possible we use standard driver-based functions for bit-banging to avoid flavour proliferation 

*    (C) we implement a standard task interface to handle the correct scheduling of callback triggers from ISR events.

*    (D) we implement the SDK required task breaks in the Wifi code.

All of this is going to involve some cod shuffling so is non-trivial. It's going to be difficult to achieve 100% backwards compatibility, but I don't really see how we can get the firmware truly stable without doing this.

I think that (C) is probably easiest to do first as a commit then (A).

jmattsson, 2016-01-12:
In regard to (A), currently it's "only" the `key` and `spi` drivers who attach ISR handlers that aren't in iram. Both `uart` and `pwm` have theirs explicitly flagged ICACHE_RAM_ATTR, and don't call anything not in iram (or ROM).

Other than that, Yes, Yes, Yes & Yes.

devsaurus, 2016-01-13:
AFAIK the ISR of the `spi` driver is exclusive to slave mode which isn't supported by the higher level at all. It's a dead stub and could be neglected if that helps anything...

devsaurus, 2016-01-13:
> "please do not occupy CPU more than 15 mSec"

Although this is not strictly related to interrupt masking - is there a way for computationally intensive code to comply with this rule *without* refactoring into timer callbacks? Eg calling a function here and there to temporarily pass CPU control back to housekeeping services?

jmattsson, 2016-01-13:
The discussion in #871 is probably relevant to that point. I'm not across co-routines, but would a yield() be useful/usable in this scenario?

TerryE, 2016-01-13:
Not quite, in `drivers` the module `readline.c` disables interrupts in `uart_getc()` in flash,  plus `spi.c` and `key.c` as you say.  The `pwm.c` routines are in RAM use their own spinlock implementation and ignore the 15 mSec guideline, so I just can't see using this PWM driver with the wifi.   `uartc` seems to be the safest driver implementation.  `i2c_master.c` looks OK.

Also `app/platform/platform.c` does tricky stuff and needs review.   The `app/modules/gpio.c` code which is enabled by `GPIO_INTERRUPT_ENABLE` is just nasty.

@devsaurus Arnim, The architectural issue is that if you break the current cb say by a task-based "yield" expecting to resume, you can't guarantee that the next task/callback is the same Lua context. You can't preempt execution within the same Lua thread.  Lua isn't architectured this way.  

As Johny suggests, Coroutines are the only way to do a long processing thread.  So the second thread would be long running and regularly yield to the foreground thread (L0) (maybe checking the elapsed time to avoid yielding too frequently).  The foreground could be a timer alarm or (when we've implemented this) a linked task cb. But

1.  Each individual cb should be less than 15mSec.
2.  The cb's as a sequence should not remain computable for more than 500mSec as this could crowd out other tasks such as network housekeeping.  If you do this the network ISRs will stack up pending context and exhaust RAM leading to a reset.  It's best to limit this b/g thread to no more than 75% of the CPU utilisation by splicing in an odd 50mSec delay every once in a while (e.g every 200mSec) to allow SDK housekeeping to flush through.

However there are two performance killers with Lua computationally intensive code: F/P emulation (if not integer build) and the current default EGC setting which triggers a GC scan on every allocation.  If you do have computational stuff to do, then it's better dropping down to C if this is time critical.

devsaurus, 2016-01-13:
@TerryE I didn't intend to question how the Lua code should be crafted. I had the code on C layer in mind.
Interface and comm modules can eat up a big share of the 15ms cake. Eg the SPI module is happily busy polling each data item while it's being transferred. In case the user asks to send 15 bytes @ 1kHz he's already in trouble. Would be great to allow for housekeeping stuff during such busy polling.
Another example are the display init sequences in u8g. Some need hundrets of ms to complete... mostly spent with waiting.
If there was a function to call in uncritical spots all these modules could keep their current architecture and just play nicely. I would hope that `os_delay_us()` served this purpose, but that's probably just wishful thinking.

TerryE, 2016-01-13:
My point is still the application programmer works at the Lua level. Let's say she has already established two Lua callbacks, a `sk.on('read', read_cb)`  and a `tmr.alarm(1,2000,1,tm1_cb)`. She then calls a long computational task which the C breaks into 15 msec chunks, say using a task cb to regain control after temporarily returning it to the SDK.

We have no way of knowing which of these three will fire next. The `on` and `tmr` callbacks will want to  return control to a Lua cb, but the computational call is blocking the running Lua thread. Lua doesn't support preemption within the same thread.  It supports multi-threading, but only through coroutines which are cooperative and not preemptive.

This is why I believe that you would need to expose this long running C task as a set of 15 msec chunks, say, at the Lua level.

TerryE, 2016-01-22:
See also #871

TerryE, 2018-06-22:
We've gone through and addressed these vulnerabilities on a point-by-point basis.  It's now time to close this catch-all Issue.

closed 2018-06-22 by TerryE