nodemcu/nodemcu-firmware #907: Strange tmr.alarm() operation

0kulist, 2016-01-06:
After hundreds of lines of code - something that drives me crazy. 
So, the story. Same problem on two devkits of type D1 mini:
1. Original NodeMCU 0.9.6 build 20150704  powered by Lua 5.1.4
2. NodeMCU custom build by frightanic.com, branch: master,	modules: node,file,gpio,wifi,net, pwm,tmr,uart, built on: 2016-01-05 16:31, powered by Lua 5.1.4 on SDK 1.4.0

I need alarm timer for periodic (up to 24h) events. So "interval", after tmr.c - uint32_t milliseconds.
Observed that example from api with a special interval fires alarm immediately - twice.

    NodeMCU 0.9.6 build 20150704  powered by Lua 5.1.4
    lua: cannot open init.lua
    > tmr.alarm(2, 11126000, 1, function() print("hello world") end )
    > hello world
    hello world

And 11126000 is only 185.43 minutes. How is that possible?

jmattsson, 2016-01-06:
I can confirm this behaviour. Something is thoroughly broken.

Not only does it fire right away, it doesn't re-fire as one would expect with a repeating timer. After testing with falling values for interval, around 6872000 I start seeing auto-repeats, but rather quick ones at that. It gets really weird once I hit 6871968, where I get autorepeat for a handful of prints, then it stops. The actual number of prints varies from run to run.
`tmr.alarm(2, 6871968, 1, function() print("hello world") end )`

`tmr.alarm(2, 6871955, 1, function() print("hello world") end )` on the other hand only gives me one print, consistently, quickly. Same all the way down to 6871949.

However, once I get to 6871948 it doesn't trigger early.

In all cases tmr.state() returns `true,1`.

Even though the interface (including in the SDK) is in ms, I believe internally the SDK uses us. Even in that scenario I'm not seeing any obvious wrap-arounds or overflows.

TerryE, 2016-01-06:
Yes but is this a bug in our firmware or in the SDK? I cant recall any special code in the firmware which could create this as we essentially wrap around the SDK timer functions. If this is an SDK bug, then should we be trying to code around it in our firmware. I think not; it would be time to claim a $1000 :)

jmattsson, 2016-01-06:
I'm poking at it... given the irregularity I'm thinking there's a good chance we're somehow clobbering the os_timer structure and thereby messing things up.

nickandrew, 2016-01-06:
Check the value of `tmr->interval` each time the timer fires?

jmattsson, 2016-01-06:
`tmr->interval` is looking good:
```
> tmr.alarm(2, 6871968, 1, function() print("hello world") end )
arming new os timer for slot 2 with interval 68dba0
> got alarm with arg 0x00000002
fielding callback on timer with interval 68dba0
recurring timer, no special handling
hello world
got alarm with arg 0x00000002
fielding callback on timer with interval 68dba0
recurring timer, no special handling
hello world
=tmr.state(2)
true	1	6871968
>
```

Moving on to the os_timer_t.

jmattsson, 2016-01-06:
Some differences:
```
> tmr.alarm(2, 6871968, 1, function() print("hello world") end )
arming new os timer 2 with interval 68dba0
> got alarm with arg 0x00000002
os_timer_t: a492fe3f7fff0f00d01800802ced264002000000
hello world
got alarm with arg 0x00000002
os_timer_t: 5c8bfe3f4f181080d01800802ced264002000000
hello world
```
Translating that into an ETSTimer struct we get
```
{
 .timer_next = a492fe3f,
 .timer_expire = 7fff0f00,
 .timer_period = d0180080,
 .timer_func = 2ced2640
 .timer_arg = 02000000
}
```
vs
```
{
 .timer_next = 5c8bfe3f
 .timer_expire = 4f181080
 .timer_period = d0180080
 .timer_func = 2ced2640
 .timer_arg = 02000000
}
```
Taking byte order into account we get the `timer_expire` fields to be 0x000fff7f vs 0x8010184f, which seems consistent with the short wait period between the first and second `hello world`. This might just be an SDK bug...

nickandrew, 2016-01-06:
What unit is `.timer_expire` in? Can you tell from low values of `interval` ?

jmattsson, 2016-01-06:
The period appears to be in units of 312500/sec.

nickandrew, 2016-01-06:
Hmm,

2^31 / 312500 * 1000 = 6871947.6736000 which aligns with your comment:

> However, once I get to 6871948 it doesn't trigger early.

6871948 * 312.5 = 0x80000066.

jmattsson, 2016-01-06:
Yeah that high bit... I'm working on a standalone test harness to see if I can reproduce this outside of NodeMCU. If I can, I'll lob the whole thing at Espressif.

jmattsson, 2016-01-06:
Alright, I've reported this to Espressif. Let's see if they come back with a fix/workaround/something.

0kulist, 2016-01-06:
Unbelievable, really nobody has faced this issue before? No one used interval longer than 2 hours?
Almost completed the project, only action left, and such a surprise.

How long can it take fixing this? Any guesstimate?
Is it worth waiting or should I try workarounds?

marcoskirsch, 2016-01-06:
You're better off working around it, with a big fat comment to come back and check if it got fixed so you can remove the workaround.

nickandrew, 2016-01-06:
@0kulist I'm not all that surprised, for something that has to happen very rarely I expect the developer will often put the device to sleep, to save power.

I suppose it's possible that an application requires the NodeMCU to be available all the time and still run a timer function only once every few hours, but obviously noone has come across this bug before.

jmattsson, 2016-01-06:
In my earlier dealings with Espressif, they've been pretty responsive. I'm hopeful I'll get some sort of answer within a few days. As @marcoskirsch said though, at this point I'd recommend working around it. One possible outcome I can see is that they classify it as a documentation error and simply update the docs rather than the code.

0kulist, 2016-01-08:
@nickandrew Do you expect a router setup page to not be available all the time ?!
Switched to rtctime, thus heavier computation under limited resources.

nickandrew, 2016-01-08:
@0kulist Router setup? Isn't that something you do once to set the SSID and password, and never need to touch again?

TerryE, 2016-01-08:
@0kulist we are taking about an issue with the firmware handling the high bit of a timer field. Let's not go down a rathole that has nothing to do with this, namely which is the best application approach to setting up the Wifi.

jmattsson, 2016-01-09:
Alright, Espressif has gotten back to me with an updated libmain.a, which appears to have fixed the issue of incorrectly fired timers. *However*, they've also clarified/reduced the maximum timeout possible in this version, which is now 0xC49BA5, which unless I'm mistaken ends up being just over 3.5hours as the longest possible timeout.

What are people's thoughts on "upgrading" to this libmain.a? Should it be added to our repo, or should we wait for 1.6.0 proper? I can obviously supply the libmain.a directly to @0kulist in that case.

TerryE, 2016-01-09:
My vote is that this isn't worth a patch upgrade. So -1 to that. As to @0kulist, if you want an arbitrary delay then you need to have a loop over alarms or deep sleep. If he or she doesn't want to do this then the source is avaible for a personal hack. But let's keep our distro clean.

pjsg, 2016-01-09:
Is nodemcu going to adopt the maximum delay from the sdk as the maximum value in the tmr.register code? Presumably the register call should fail if the supplied value is too large. It seems that either it fails or there needs to be an emulation layer that converts very long delays into multiple shorter ones. I think that that is the path to madness... (and code bloat).

0kulist, 2016-01-09:
Thank you, @jmattsson, for update on this issue. As about me - in this circumstances, I'll stick with mentioned rtctime. However, it would be nice to have this update available to everyone. Also, it would be nice to have documentation updated, not to mention that this saving rtctime was found only by tmr.alarm() accident (bug). 
Also I vote, if taken into account, for extending tmr.alarm interval to fully support declared uint32_t milliseconds the next version. (Didn't realize why nickandrew mentioned 2^31)

TerryE, 2016-01-09:
@0kulist, unfortunately not all votes are equal. Those who are actively engaged in the development of the firmware and proactive bug fixing have earned more karma to influence decisions.

nickandrew, 2016-01-09:
I guess the only benefit of not using the 1.5.0 -dl branch is that it's possible to patch the SDK between releases. However, the SDK releases seem to be fairly frequent, so there's probably not that long to wait for a 1.6.0.

How about stick with the known limitations of 1.5.0 for nodemcu right now, and then raise that value to 0xC49BA5 when 1.6.0 is integrated.

Also, do we have any idea how 0xC49BA5 is calculated? It used to be 0xFFFFFFF (only 7 nibbles long!) What limitation is coming into play here? It looks like the requested interval (in us) is multiplied by 312.5, being some sort of more fundamental unit which the hardware is counting, but 3.5 hours is nowhere near the 32 bit uint capacity.

jmattsson, 2016-01-09:
There was no mention of the reason behind 0xC49BA5, and I couldn't come up with any multiplier that ended up on an interesting bit boundary.

@nickandrew We could still patch the SDK with the -dl approach. I'd simply upload the new libmain zip somewhere accessible, and make it part of the download & extract targets.

I don't mind waiting for 1.6.0, especially given the fixed version doesn't extend the working range by that much anyway. I've got a local patch to the tmr module that range checks the timeout value, I'll clean that up and submit a PR for that at least. And I or someone else should make sure the range is documented in the right spot in the `newdocs` branch for when we switch over. Feel free to remind me if I haven't updated that by the time I PR the tmr patch.

lepass7, 2016-01-10:
@0kulist 
I had the same problem and I solve this with a workaround:
MAX_TIME = 24
tmr.alarm(0, 60*60*1000, 1, function()
timer = timer + 1
if (timer >= MAX_TIME) then
  --timer interval gone
  --DO here what you want
  tmr.stop(0)
  timer = 0
end
end)

TerryE, 2016-01-10:
@lepass7,  thanks for your example, but can I suggest that you wrap this logic in a function, say longTimeAlarm(), use a closure upval for the long time count down and unregister() rather than stop() to prevent retaining the callback reference in the Lua registry.

0kulist, 2016-01-10:
@lepass7 @TerryE 
And if you need minute or second resolution this tend to get more complicated, taking into consideration limited resources.

TerryE, 2016-01-10:
@lepass7, but using locals and unregister() take _less_ resources not more -- but explaining how to write efficient Lua is a #719 issue and is not for here. 

Given #916, we can close this issue.

closed 2016-01-10 by TerryE

pjsg, 2016-01-10:
We are still waiting for @jmattsson PR to add the error checks (at least).......

TerryE, 2016-01-10:
@pjsg ? #916 ?

pjsg, 2016-01-10:
That will teach me to post before coffee..... Sorry.

TerryE, 2016-01-10:
:wink:

barbonist, 2016-01-11:
hi all , i just started coding in lua for nodemcu. I see strange behaviour when using the tmr.alarm(

This is my code snippet
ison = true;
 gpio.mode(10,gpio.OUTPUT)
 tmr.alarm(1, 100, 1, function()
	if(ison) then
		gpio.write(10,gpio.LOW)
		ison = false;
		print("off ")
	else
		gpio.write(10,gpio.HIGH)
		ison = true;
		print("on ")
	end
 end)

The led blinks as expected , but i cannot see the on / off messages  does'nt seem to exit on the usb (  i am verifying on teraterm ).  Any idea ? 
Thanks a lot
bye Stefano

TerryE, 2016-01-11:
@barbonist Stefano, please see our guidance at #719.

stevemac00, 2016-05-02:
This bug was annoying but with considerable amount of code (relatively speaking) I was able to work around. If it can't be fixed, can we correctly document it? The current documentation about the bug appears to be wrong. I.E.
> Maximum value is 12884901. In SDKs <= 1.5.0 values >6871948 result in incorrect behaviour

However, I just built with latest nodemcu-firmware-dev1.5.1 with
`git clone --recursive https://github.com/pfalcon/esp-open-sdk.git`
and this still fails:
`tmr.alarm(0,12600000,0,function() print("\nDone\n") end)--still fails in dev branch 1.5.1`
which if multiplied by a 1000 would have bit 31 set.