nodemcu/nodemcu-firmware #911: ADC read wrong value

TheB1ackSheep, 2016-01-07:
I just build firmware from http://nodemcu-build.com/ with adc module and it is always return 65535.

tertuis, 2016-01-07:
I'm having the same issue, tried both the master and dev branches and both read 65535, even if the pin is grounded.

tertuis, 2016-01-07:
There's something weird going on here....

When I use adc.readvdd33() then it appears to actually read the ADC pin, although the rails are incorrect. The range is from 26 to 4094, for a dead short and full 3.3V applied to pin (I realise 1V is the max.).

See if that also works for you?

eyaleb, 2016-01-07:
This is not a bug, but a "feature" of the SDK, where it says to leave `TOUT` unconnected to read Vdd. However I do not know how to set the config to read `TOUT` in `nodemcu-build.com`.

If you build at home: to read `TOUT` with `adc.read(0)` you need to select it in `app/include/user_config.h`:
```lua
// Byte 107 of esp_init_data_default, only one of these 3 can be picked
//#define ESP_INIT_DATA_ENABLE_READVDD33
#define ESP_INIT_DATA_ENABLE_READADC
//#define ESP_INIT_DATA_FIXED_VDD33_VALUE 33
```
You will not be able to use `adc.readvdd33()`.

tertuis, 2016-01-07:
Understood.

However reading the 3.3V supply actually gives me the TOUT value. In the printout below "Water_Sensor" is a normalised version of adc.readvdd33() and tracks the ADC (TOUT) pin correctly, not the supply rails:

Temperature: 24 deg C
Humidity: 36%
Water Sensor: 0%
***Wifi status: 1 
Temperature: 24 deg C
Humidity: 36%
Water Sensor: 28%
***Wifi status: 1 
Temperature: 24 deg C
Humidity: 36%
Water Sensor: 0%

marcelstoer, 2016-01-07:
> I do not know how to set the config to read `TOUT` in nodemcu-build.com.

It's not supported. You may want to try a custom build using the Docker image: https://github.com/marcelstoer/docker-nodemcu-build.

fixingthingsguy, 2016-01-08:
My issue is similar but a bit confounding. I get different analog values with different hardware using a standard "frightanic dev or master build".
NODEMCU dev board with AI thinker written on it       - adc.read(0) works fine
NODEMCU dev.board with nameless on board            -adc.read(0) gives 65535 which is an error I believe.
I want to use the same build, so its a bit of a problem if I have to use different builds to secure ADC(Tout). Not sure if there is a way out. I did try to make a custom build, but as, they say, "lord have mercy". Its not for everybody! I'm a novice kind of guy, no make/build knowledge , and the myriad steps, specifity of the recipe, etc, is such that  to make a custom build is not for me, now.
 I just hope the community will come up with some creative way of not having to use custom builds for 
nuts and bolts operations, such as using an ADC. Is there a simple hardware fix that can work on the one board, and not on the other?

mvilla14, 2016-01-31:
I have the same problem reading with the function ​​adc.read(0) always returns 65535 , I using the SDK 1.5.1 floating.bin ( created with cloud build service ) , but it also happens with SDK 0.9.6 floating.bin, strangely it was in operation properly with both SDK, but at a certain point began to fail and return 65535. What would be the solution ? . The adc.readvdd33 () function returns a value of approximately Vdd.

jmattsson, 2016-02-01:
@mvilla14 Per [the docs](https://nodemcu.readthedocs.org/en/dev/en/modules/adc/), use the Espressif flash download tool to write a different esp_init_data block to change the value of byte 107.

mvilla14, 2016-02-01:
It solved perfectly , thanks.

TheB1ackSheep, 2016-02-01:
@mvilla14 How did you solved it?

mvilla14, 2016-02-01:
I edit the 'Esp_init_data_default.bin' file, and at position 107, I put the
value 33 which is corresponding to a supply voltage of 3.3V, thus reading
is enabled 'adc.read(0)'.

n8twj, 2016-02-01:
Could someone explain the process of generating the Init Bin using Flash download tool?   both settings i have tried adc.read(0) always returns 65535, but i am not sure I am doing it correctly. 

$ python ./esptool.py --port=/dev/tty.wchusbserial1410 write_flash -fm=dio -fs=32m 0x7C000 ~/Development/esp/esp_init_data_default.bin 
Connecting...
Erasing flash...
Wrote 1024 bytes at 0x0007c000 in 0.1 seconds (68.7 kbit/s)...

Also, building via the Docker image, I have tried all three values (one at a time) in app/include/user_config.h and adc.read(0) always returns 65535    :(

$ python ./esptool.py --port=/dev/tty.wchusbserial1410 write_flash -fm=dio -fs=32m 0x00000 ~/Development/nodemcu-firmware/bin/nodemcu_float_master_20160201-2011.bin 
Connecting...
Erasing flash...
Wrote 477184 bytes at 0x00000000 in 56.8 seconds (67.2 kbit/s)...

Leaving...

from Esplorer:
=adc.read(0)
65535

devsaurus, 2016-02-01:
> but i am not sure I am doing it correctly.
> 
> $ python ./esptool.py --port=/dev/tty.wchusbserial1410 write_flash -fm=dio -fs=32m **0x7C000**

Correct location for the init data is `0x3fc000` on a 4&nbsp;MByte flash: Chapter 3 in [2A_ESP8266_SDK Getting_Started_Guide_EN_v1.5_20151118.pdf](http://bbs.espressif.com/viewtopic.php?f=51&t=1024)

n8twj, 2016-02-01:
This works... Thank you!    I edited byte 107 (as documented) and wrote to nodemcu like this: 

$ python ./esptool.py --port=/dev/tty.wchusbserial1410 write_flash -fm=dio -fs=32m 0x3fc000 ~/Downloads/esp_init_data_default.bin 
Connecting...
Erasing flash...
Wrote 1024 bytes at 0x003fc000 in 0.1 seconds (68.7 kbit/s)...

> =adc.read(0)
181     <---  moisture sensor in wet soil
> =adc.read(0)
634    <---  moisture sensor pulled out (open air)

closed 2016-02-02 by TerryE

fixingthingsguy, 2016-02-02:
Thank you community! This is a merciful way rather than doing a rebuild. An uber merciful way would be to get  Expressif to change the default to ADC(Tout). It would distinguish their device from those that do not have ADC(Tout) and would enable a lot more people to use it.

ybycode, 2016-03-20:
Thanks for this.
For those who'd like to change the esp_init_data_default.bin file and can't use the Espressif flash download tool (which is for MS Windows only), here is the command line that does the work:

`printf '\x21' | dd of=esp_init_data_default.bin bs=1 seek=107 count=1 conv=notrunc`

It worked for me, after flashing the file with the same command as @n8twj:

`./esptool.py --port=/dev/ttyUSB0 write_flash -fm=dio -fs=32m 0x3fc000 esp_init_data_default.bin`

gkzsolt, 2016-03-26:
Thanks, ybycode, I have the same problems and tried this way. However, I get the following error on write_flash:
`$ esptool.py --port=/dev/ttyUSB0 write_flash -fm=dio -fs=32m 0x3fc000 esp_init_data_default.bin
Connecting...
Erasing flash...
Took 0.01s to erase flash block

A fatal error occurred: Failed to enter Flash download mode (result "0x1, 0x6")`

I got esp_init_data_default.bin from [https://github.com/espressif/ESP8266_AT/tree/master/bin](here) (one year old, is it good?). Also, I am trying to flash over the nodemcu build, if this is all right. No luck, however.

gkzsolt, 2016-03-26:
Ok, solved, by getting the last firmware from Espressif. [This](http://www.allaboutcircuits.com/projects/update-the-firmware-in-your-esp8266-wi-fi-module/) post helped me navigate through the not-so-nice chinese site ;)

After writing 33 to address 107 adc.read(0) seems to work, although not quite in sync with my multimeter;) I wonder if anybody did an evaluation of their ADC?

Wifsimster, 2016-07-01:
If someone has the nodemcu-build.bin with the correct fix, i'm looking for it.

Thanks.

marcelstoer, 2016-07-01:
a) this is not a support forum -> http://nodemcu.readthedocs.io/en/latest/en/support/
b) ADC can be set in Lua at runtime -> http://nodemcu.readthedocs.io/en/latest/en/modules/adc/#adcforce_init_mode
c) you need an up-to-date firmware (1.5.1) for that -> http://nodemcu.readthedocs.io/en/latest/en/build/