nodemcu/nodemcu-firmware #913: socket.on:recieve sometimes not working

Hellsy22, 2016-01-08:
I have my own http-client lua library. It's working well with a dev96 Aug 2015 firmware.
But with fresh firmware (tried both master and dev) I got strange bug - event "receive" never happens in module, but working if I send commands one by one with LuaLoader. Same if I tried to send all commands in one string.

One string. No output
```lua
sk=net.createConnection(net.TCP, 0); sk:on("receive", function(sck, c) print(c) end); sk:connect(80, "192.168.0.1"); sk:send("GET /192.168.0.1.lua HTTP/1.1\r\nHost: esp\r\nConnection: close\r\nAccept: */*\r\n\r\n")
```

Function. No output
```lua
function justatest()
   sk=net.createConnection(net.TCP, 0);
   sk:on("receive", function(sck, c) print(c) end);
   sk:connect(80, "192.168.0.1");
   sk:send("GET /192.168.0.1.lua HTTP/1.1\r\nHost: esp\r\nConnection: close\r\nAccept: */*\r\n\r\n")
end
justatest()
```

Two strings with 1 sec pause. Page printed
```lua
sk=net.createConnection(net.TCP, 0); sk:on("receive", function(sck, c) print(c) end); sk:connect(80, "192.168.0.1")
sk:send("GET /192.168.0.1.lua HTTP/1.1\r\nHost: esp\r\nConnection: close\r\nAccept: */*\r\n\r\n")
```

I tried to use "connection" event and it happend, but nothing changes.
I tried to delay script execution by tmr.alarm(...), it not helped.

TerryE, 2016-01-08:
You have to do the sk:send() in an sk:on('connect', callback) with SDKs after V1.0. 

But this issue falls under #719

closed 2016-01-08 by TerryE

Hellsy22, 2016-01-21:
Finaly I got it.

```lua
url = "/"
host = "api.ipify.org"
port = 80
ip = "54.225.138.188"

sk=net.createConnection(net.TCP, 0)
sk:on("connection", function (sck)
	sck:send("GET "..url.." HTTP/1.1\r\nHost: "..host.."\r\nConnection: close\r\nAccept: */*\r\n")
	sck:send("\r\n")
end)

sk:on("disconnection", function(sck) print "Disconnected\n" end);
sk:on("receive", function(sck, c)
	print("got: " .. c)
end)

sk:connect(port, ip)
```

All ok if only one send used:
```
sck:send("GET "..url.." HTTP/1.1\r\nHost: "..host.."\r\nConnection: close\r\nAccept: */*\r\n\r\n")
```

Not working with two consequent send:
```
sck:send("GET "..url.." HTTP/1.1\r\nHost: "..host.."\r\nConnection: close\r\nAccept: */*\r\n")
sck:send("\r\n")
```