nodemcu/nodemcu-firmware #915: multiple file.write cmd break spiffs

danlu01, 2016-01-10:
I'm using the latest master build firmware.

When I'm writing a file to esp8266, after a while it would rst and no longer able to write or delete files.  I would need to issue file.format() cmd to fix the spiff system.  Basically, with trail and error, I think the cause is when the previous file write cmd didn't finish before the next file write cmd.  Is there a why we can add a hook to check when the file.write cmd is done?

Test:
Loop this
file.open("testfile","w+")
file.write(str1)
file.write(str2)
file.write(str3)
file.write(str4)
file.write(str5)
etc.

After a few times the above will reset and won't be able to create new files or delete old ones.

If I added some delays between file.write, the problem occurs less frequent.

danlu01, 2016-01-11:
Did more testing.. I think the issue is with the SPIFF system itself.  After X amount of write to the flash, it gets corrupted.  I tested by adding file.format() between each test and was able to loop overnight without problems.  However, this is very risky solution to use in the field as power outage will render the device useless.

TerryE, 2016-01-11:
Out of interest, `file.write()` returns a status.  Why aren't you checking this if you are having write problems?  Also writes take time and there is a general time limit for any callback.  Please put your entire test code in a gist and provide a link so we can see what you are doing.

danlu01, 2016-01-11:
Once it's corrupted, any file.write() causes an immediate reset/crash so there's no way to catch the error.

Here is my test loop file, it prints number of time it have run.  Usually takes 150+ iterations before SPIFF is corrupted.  Once corrupted, file.remove or file open with overwrite will cause immediate rst error.  However, if it will let you file open with append and any file.write issues is reported successful but file.close() will cause immediate rst but no modification is saved.

https://gist.github.com/danlu01/4d06ec4e1b7836f5a246

danlu01, 2016-01-11:
ok, I just noticed it always crashes on the 185 files on a fresh file.format() run.  Is this bug because it's allocating new flash address each time even though it's just writing to the same file?

182
183
184
185
bBÊÇþ¤„ÿ
æÇ!„þ1ä)Í!ÜÎÿÐCá
NodeMCU custom build by frightanic.com
	branch: master
	commit: c8037568571edb5c568c2f8231e4f8ce0683b883
	SSL: false
	modules: node,file,gpio,wifi,net,pwm,tmr,uart
 build 	built on: 2016-01-10 00:48
 powered by Lua 5.1.4 on SDK 1.4.0
lua: cannot open init.lua
> 

Once it reboots, FS info reports:
Total : 565001 bytes
Used  : 2259 bytes
Remain: 562742 bytes

danlu01, 2016-01-11:
Is there a way to store some LUA codes in ROM or limit the section of flash memory for file.formatI()?  This way I can store some LUA codes as backup in case file.format() is needed as workaround for above?

TerryE, 2016-01-11:
I use my own simple provisioning system where a small bootstrap establishes a UDP session with a provisioning server on my network (an RPi2). This bootstrap formats the SPIFFS and the downloads a fresh copy of the app onto the file system before rebooting.    Some day, I intend to turn this into an example, once I get a breathing space.  In the meantime, thanks for the gist.  We'll take a look at this.

Alkorin, 2016-01-11:
Tested on mine, it brokes after 240 writes

```
236
237
238
239
240
PANIC: unprotected error in call to Lua API (stdin:6: open a file first)
€€

NodeMCU 1.5.0 build unspecified powered by Lua 5.1.4 on SDK 1.5.0
```

TerryE, 2016-01-11:
Bugger!

Alkorin, 2016-01-11:
I've added some debug, and now it doesn't crash 750 writes and counting -_-
This bug will be hard to hunt :/

TerryE, 2016-01-11:
Are you clearing the WDT?

Alkorin, 2016-01-11:
It's not a watchdog issue on mine, I've added error handing on @danlu01 gist and nodemcu survives ...
=> https://gist.github.com/Alkorin/dc667dcbbe0d4999aabf

```lua
[...]
true
52
true
53
true
54
true
55
-1
DEAD
> print(file.open("user.lua","w+"))
-1
> print(file.open("user2.lua","w+"))
-1
> file.format()
format done.
> print(file.open("user.lua","w+"))
true
```

As you can see, I stop the loop on file.open error. After this error, any call to "file.open" will return an error and file.format() fixes the case...

Alkorin, 2016-01-11:
For posterity:

```
append: 8002 store modified objix_hdr page, 0022:0000, written 242
append: 242 bytes @ offs 972 of size 972
append: gc check fail -10002
append: 66 bytes @ offs 1214 of size 972
append: offset reversed to size
append: gc check fail -10002
```

Alkorin, 2016-01-11:
@jmattsson any idea ? :)

TerryE, 2016-01-12:
This kind of big needs ruminating over. It will be something subtle, an edge case in the GC, IMO

jmattsson, 2016-01-12:
Hmm. On e9f16d1ace183bedfb045a5c2681b3a2a6df360e I could kind-of reproduce it. My devkit0.9 board would reset after iteration 156, reliably, and present reset cause 2 (software reset). Wanting to get a bit more info, I added the system_get_rst_info() to node.bootreason() and rebuilt + reflashed. The problem then moved (yay, not...). I seem to be hitting a crash at one of three locations:

* 2x Crash observed after iteration 441, allegedly in ieee80211_add_xrates() for address 0x19. Seems like a null-pointer dereference. Filesystem was usable afterwards.

* 2x Crash after iteration 225, allegedly in spiffs_gc_find_candidate() on access 82 bytes into spiffs_work_buf . Filesystem was **not** usable afterwards, *once*. The other time is was hyper-usable and the test happily ran past interaction 666 (see next item).

* 1x Crash after iteration 666, allegedly also in spiffs_gc_find_candidate() on access 82 bytes into spiffs_work_buf . Filesystem **not** usable afterwards. I'm dumping the resulting file system now for a post-mortem.

* 3x Crash after iteration 440, allegedly in esf_alloc_buf() accessing address 0. Filesystem was usable afterwards.

Worth noting is that my ESP was connected to a relatively busy WiFi network at the time.

jmattsson, 2016-01-12:
Okay, two separate issues here:
  1. The test gist causes the heap to run out due to receiving packets and not processing them, causing various crashes in the network stack related to failed allocations. Changing from tmr.ALARM_AUTO to tmr.ALARM_SEMI plus a tmr.start(5) at the end of the callback gives the network stack a chance process packets and not run the system out of memory.
  1. The actual SPIFFS bug is [here](https://github.com/nodemcu/nodemcu-firmware/blob/master/app/spiffs/spiffs_gc.c#L254). Note the "helpful" comment '// todo alignment?'... I'll prepare a workaround patch for NodeMCU for now, and report upstream time permitting. Someone else welcome to beat me to that though.

Alkorin, 2016-01-12:
Hum @jmattsson on mine it doesn't reboot, which gist did you used ?

I've upgrades spiffs to 0.3.3 which fix alignment
```
diff --git a/app/spiffs/spiffs_gc.c b/app/spiffs/spiffs_gc.c
index a86f812..a0837c3 100644
--- a/app/spiffs/spiffs_gc.c
+++ b/app/spiffs/spiffs_gc.c
@@ -36,7 +36,7 @@ s32_t spiffs_gc_quick(
   int cur_entry = 0;
   spiffs_obj_id *obj_lu_buf = (spiffs_obj_id *)fs->lu_work;
 
-  SPIFFS_GC_DBG("gc_quick: running\n", cur_block);
+  SPIFFS_GC_DBG("gc_quick: running\n");
 #if SPIFFS_GC_STATS
   fs->stats_gc_runs++;
 #endif
@@ -249,10 +249,15 @@ s32_t spiffs_gc_find_candidate(
   c_memset(fs->work, 0xff, SPIFFS_CFG_LOG_PAGE_SZ(fs));
 
   // divide up work area into block indices and scores
-  // todo alignment?
   spiffs_block_ix *cand_blocks = (spiffs_block_ix *)fs->work;
   s32_t *cand_scores = (s32_t *)(fs->work + max_candidates * sizeof(spiffs_block_ix));
 
+   // align cand_scores on s32_t boundary
+#pragma GCC diagnostic push
+#pragma GCC diagnostic ignored "-Wpointer-to-int-cast"
+  cand_scores = (s32_t*)(((ptrdiff_t)cand_scores + sizeof(ptrdiff_t) - 1) & ~(sizeof(ptrdiff_t) - 1));
+#pragma GCC diagnostic pop
+
```

But still have the same issue :(

jmattsson, 2016-01-12:
@Alkorin I used your gist. Like I mentioned though, it's on a fairly busy network which would make a big difference. I printed node.heap() in each iteration, and it was steadily going down.

Could you try my workaround patch rather than the 0.3.3 upgrade? That might be dying in another place. Squash one bug at a time and all that...

Alkorin, 2016-01-12:
Will do @jmattsson, but this evening (~21:00 GMT+1 ;))

TerryE, 2016-01-12:
> The test gist causes the heap to run out due to receiving packets and not processing them, causing various crashes in the network stack related to failed allocations. Changing from tmr.ALARM_AUTO to [...]

@jmattsson, Johny.  Not directly related to this issue, but this lack of understanding that the SDK really masks what is a form of cooperative muli-tasking so if you write your app to mask interrupts for too long or simply run your callbacks for too long so that other tasks (often but not always related to serving the network stacks) can't clear their workload then bad things start to happen is really not well understood.  We could really do with a FAQ topic on this in its own right.

jmattsson, 2016-01-12:
The risk of creating a "backlog" of expired timers probably warrants an inclusion in the new API docs, I agree. Not that it's mentioned in the official SDK docs either, but now that we're aware of it we should make note of it at least.

I'm half asleep at the moment so I'm not doing it right now. Remind me if I forget (or beat me to it so I don't have to do it)

TerryE, 2016-01-12:
Not saying that you should do it, but the reality at the moment is that it will probably end up as one of us two until the other contributors climb this learning curve, which some seem to be doing :smile:

jmattsson, 2016-01-14:
@Alkorin have you had a chance to do some more tests by any chance?

danlu01, 2016-01-20:
Any head way on flash writing bug?  When I writing to single file over and over again vs writing to new files, the failure happens at the same iteration.  Could this bug be just an address allocation issue?

jmattsson, 2016-01-20:
@TerryE can I suggest we merge in that ugly fix for now as a stop-gap measure? I don't think I'll get a chance to try an upgrade to 0.3.3 and debug that in the near future.

@danlu01 nope, it's an alignment issue in the SPIFFS garbage collector, details [here](https://github.com/nodemcu/nodemcu-firmware/issues/915#issuecomment-170810675).

danlu01, 2016-01-20:
@jmattsson but Alkorin mentioned he tried the 0.3.3 spiffs with alignment fix but still have same issue.

jmattsson, 2016-01-20:
@danlu01 Yes, but we don't yet know what the failure point is in 0.3.3. I suspect there has been a regression somewhere else there.

With our existing 0.3.2 plus the ugly fix, I can keep rewriting a file. Without the ugly fix, I cannot.

Once we upgrade to 0.3.3 we'd need to spend time debugging it further though.

danlu01, 2016-01-20:
@jmattsson Can you implement the ugly fix to existing 0.3.2 to Master or Dev so I can use it to build a custom firmware with http://nodemcu-build.com/index.php.

Thanks, Daniel

jmattsson, 2016-01-20:
@danlu01 Terry already merged it, it's in `dev` now. Let us know if it's still breaking for you after that.

danlu01, 2016-01-20:
ok, I'll try now.. thanks guys!!

PR #923 (2016-01-20): Ugly workaround for broken SPIFFS GC alignment.
This fixes #915, for now.
closed 2016-01-20 by TerryE