nodemcu/nodemcu-firmware #918: tmr.alarm memory leakage

lepass7, 2016-01-10:
I have noticed that tmr.alarm has memory leakages and also tmr.stop wont release memory used by tmr.alarm.
After restart:
node.heap() = 41720
calling tmr.alarm
node.heap() = 41040
fires tmr.alarm
node.heap() = 41480

So by just calling a timer I have lost 240 bytes of ram.

I have also noticed that tmr.stop wont release memory that tmr.alarm needs. For example
After restart:
node.heap() = 41720
calling tmr.alarm
node.heap() = 41040
tmr.stop
node.heap() = 41040

So  I have lost 680 bytes of ram. 

I think this is a bug

TerryE, 2016-01-10:
See #382 and #413.

closed 2016-01-10 by TerryE

jmattsson, 2016-01-11:
To release the resources you also need to `tmr.unregister()` in many cases. See the [updated docs](http://nodemcu.readthedocs.org/en/newdocs/en/modules/tmr/).