nodemcu/nodemcu-firmware #920: TM1829 module missing in master & dev

marcelstoer, 2016-01-11:
Just noticed that #686 is missing in master & dev. I couldn't find any notes as to why this could be intentional. Was it simply forgotten in the recent 1.4 upgrade and the necessary branch/PR clean-up (PR was directly merged to master)?

jmattsson, 2016-01-12:
I'm not aware of any reason why it should be excluded. I'd say it was simply missed in the cleanup.

TerryE, 2016-01-12:
+1 My bad.

devsaurus, 2016-01-12:
Thought it was still awaiting some kind of check/verification since it never got in touch with a recent SDK on `dev`.

TerryE, 2016-01-12:
Yup, I did ask the contributors to resubmit, but I should have followed it up.

marcelstoer, 2016-01-16:
@h3ndrik can you please take care of that.

h3ndrik, 2016-01-28:
thanks. submitted new pull requests for master #983 and dev-branch #984

closed 2016-01-28 by marcelstoer