nodemcu/nodemcu-firmware #921: dht.c: do not source the line

dvv, 2016-01-12:
https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/dhtlib/dht.c#L310-L313

Please consider commenting these lines out so the pin stayed Z-input (or better internal pullup) input. The task of recharging the idle signal line should be left to an external pullup.

TIA

pjsg, 2016-01-12:
I would be unhappy if this incompatible change was made. It would require a *hardware* change to existing implementations to keep things working. It might be possible to make this behaviour a non-default option.

dvv, 2016-01-12:
I see. Providing one runs nodemcu all he needs is to manually set the mode and value after DHT read is done -- so no hardware changes, only software small ones.
This is what I do right now (only I do set the pin to input) to cope with the issue.

Personally, I consider the following schema: for short distance to DHT internal pullup suffices, for long distance only via external pullup the line capacity can be recharged safely without exceeding pin max power.

Moreover, I believe not many people ever noticed that pin they assume steady input does source in fact, even at idle. It is just unsafe.

An option strike the ease of use of nodemcu-build.

nickandrew, 2016-02-10:
Could disabling output mode be added as an option?

Alternately, would sticking a 220Ω resistor inline with the GPIO pin satisfy your current drain concerns?

pjsg, 2016-02-10:
Looking at it again, I think that you are right -- the data pin should switch between driving LOW and being an input with pullup.

closed 2019-06-21 by stale[bot]