nodemcu/nodemcu-firmware #925: Implement coder module

TerryE, 2016-01-12:
Implement `coder` module with methods:

*    toBase64
*    fromBase64
*    toHex
*    fromHex

Denormalise `crypto` to call the coder.toXXX routines.  Discussion on #486 refers

marcelstoer, 2016-01-12:
> Denormalise `crypto` to call the coder.toXXX routines.

Not sure I understand.

Would that be the first time that two non-core modules actually depend on each other (see https://github.com/nodemcu/nodemcu-firmware/issues/386#issuecomment-115108172)? Why wouldn't you want to remove the methods in `crypto`? Calling the old `crypto.xxx` functions would still fail on a firmware that doesn't have the `coder` module, right?

jmattsson, 2016-01-12:
Hmm, that's a good point. I hadn't thought of that added complication for the backwards compatibiilty. Maybe that's enough to not do it and "fix it in documentation" instead? From a user perspective I'd still prefer backwards compatibility, but I definitely hear you on the technical difficulties.

Maybe we should start out with an
   ```
#if defined(LUA_USE_MODULES_CRYPTO) && !defined(LUA_USE_MODULES_CODER)
# define LUA_USE_MODULES_CODER
#endif
```
and then officially split them a bit later down the track when we feel we have time to maintain "stuff that has changed from version x to y that you need to be aware of" notes?

marcelstoer, 2016-01-12:
A software that is properly released usually has some deprecation policy based on a release plan in place. We don't have that and would have to rely on time-based deprecation. That would/could mean we create the new module, *copy* the methods and add a deprecation note to the docs for `crypto`. It could state that the deprecated methods are removed in n weeks or months.

TerryE, 2016-01-12:
I can leave the old code in crypto with a  `#ifdef LUA_USE_MODULES_CODER` option to substitute the the calls to coder instead.  But I won't add any new methods to crypto.

Alkorin, 2016-01-18:
`encoding` isn't a better name than `coder` ?
It's the name of golang's package for this kind of stuff

https://golang.org/pkg/encoding/hex/
https://golang.org/pkg/encoding/base64/

TerryE, 2016-03-02:
Done

closed 2016-03-02 by TerryE

devyte, 2016-03-03:
@TerryE Thank you for this, I _really_ want to get rid of the lua implementation of fromBase64() in my code.
@marcelstoer Is it possible to add the encoder module to the nodemcu firmware build service?

marcelstoer, 2016-03-03:
Yes, it is possible and sooner or later I will. There are a number of other modules that were merged into the `dev` branch recently which are not yet integrated into the cloud builder.

jmattsson, 2016-03-03:
One of these days we'll get a chance to rework how the whole module inclusion system works, and hopefully to something the cloud builder automatically picks up on.