nodemcu/nodemcu-firmware #926: ow: allow pin 0

dvv, 2016-01-12:
https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/modules/ow.c#L14

The condition forbids using 1-wire on pin 0.
Is it intended?

TIA

ABonner, 2016-01-22:
Isn't GPIO0 used to select bootup mode (Boot from Flash or Program via UART)?

dvv, 2016-01-22:
Right. But nodemcu is master and OW devices never (afaik) lower the line without master command and line is pulled up.

nickandrew, 2016-01-31:
D0 is GPIO16, which has restricted functionality (no interrupts or PWM). I don't think there's any reason that the ow module *as currently designed* would not work with GPIO16. However, if the ow module were changed to be interrupt-driven (which I'd recommend) then D0 would become unavailable again.

Is there a need for you to run onewire on GPIO16? You're using every other GPIO already and none of *those peripherals* can use GPIO16?

dvv, 2016-01-31:
Let us count it as a notice. +1 for interrupt-driven ow.

nickandrew, 2016-02-01:
We're agreed that an interrupt-driven onewire module would be a good thing. It just needs someone to take the time to code it. And if it happens, D0 won't be usable anyway. So there's no point enabling D0 in the current module. Closing.

closed 2016-02-01 by nickandrew

dvv, 2016-03-22:
I wonder if UART as 1-wire driver (https://www.maximintegrated.com/en/app-notes/index.mvp/id/214) can be used for interrupt-driven onewire module?

nickandrew, 2016-03-22:
@dvv I think it's feasible, so long as the hardware designer doesn't mind using 2 pins for each 1-wire bus, and specifically 2 UART pins. They also have to make an external open-drain buffer circuit to connect the ESP-xx to the onewire bus.