nodemcu/nodemcu-firmware #927: Switch on more compiler warnings; improve codebase accordingly

jmattsson, 2016-01-12:
Currently we're building NodeMCU with the inherited compiler settings from Espressif. They're very, very lax, and not doing much to help us get a high quality code base. While [some code](https://github.com/nodemcu/nodemcu-firmware/blame/master/app/modules/tmr.c#L1) is compiled with higher warning levels, most isn't.

Getting to -Wall -Wextra with -std=gnu99 or gnu11 for app/modules, app/platform and app/user should be our minimum aim. Third party code we clearly have less influence over, but at the very least we should build that with their recommended warnings enabled.

I'm hopeful that getting us standards clean will also improve the code generation and thus give us a reduced flash foot print.

devsaurus, 2016-01-13:
Fully agree and consider me in for any overhaul of the code base.
I've been bitten a couple of times just because the compiler silently accepted crappy code in the first place.

TerryE, 2016-01-13:
+1, but its going to be a pain in the arse clearing all of those warnings.  Divide and conquer?

devsaurus, 2016-01-21:
> Divide and conquer?

We had good experience with this approach for the doc transfer job. Ground rules were:
* central "job table" [Module list](https://github.com/nodemcu/nodemcu-firmware/wiki/Module-list) where WIP status is published to avoid duplicate / conflicting efforts
* direct commits to a dedicated branch
* have an issue for discussion

Looking at #958 the division lines would be defined by the Makefiles, further boiling down to directories or even whole subtrees.
@jmattsson how about a similar approach? Set up a wiki page containing all "Makefiles" to take away for WIP and check mark when done. Commits go into a dedicated branch at nodemcu-firmware repo (*insert snappy name here*) until it's finally merged to `dev` - Or do you prefer a PR per developer and merge these one by one?
I would set up branch and wiki page if you're ok with that.

jmattsson, 2016-01-21:
I would really prefer we get incremental improvements into `dev` as we go along. If we don't do that we'll discover by the time we're ready to merge back that there's another ton of things that need to be fixed that have crept in in the meantime.

We don't need a big-bang approach here, and in fact it might be outright helpful to do this piecemeal. There will be a substantial amount of changes resulting from this, and if we get testing on each chunk as we go along it will be a lot easier to pinpoint any regressions that might be introduced.

If people are interested in working on this, having a "job table" on the wiki could be handy.  Something like this perhaps:

Dir | gnu11 | Wall | Wextra | notes
----|------|------|----|-----
app/modules | Johny | | |
app/user | | | |
app/platform | | | |
app/driver ||||
app/crypto ||||

Dir | third-party cflags | notes
----|----|---
app/http ||
app/mqtt ||
app/lua ||
app/json || (is this even used? if not, can/should we nuke it?)
app/tsl2561 ||
app/u8glib ||
app/ucglib ||
app/wofs || (should we keep this support around any longer?)
app/upgrade || (I believe this dir should be dropped, it's old SDK upgrade files)

TerryE, 2016-01-21:
I'd certainly vote for dropping **upgrade** and **wofs**

devsaurus, 2016-01-21:
Also :+1: for dropping wofs. It doesn't look like a drop-in alternative to spiffs anymore.

marcelstoer, 2016-08-07:
I closed it because we've had the job table for months. I don't see merit in keeping this open until all issues the initial description implicitly includes are fixed.

closed 2016-08-07 by marcelstoer