nodemcu/nodemcu-firmware #929: Bug in rotable handling

TerryE, 2016-01-13:
This is an elua bug but I'll post it here first.  There are two parts (i) a bug and (ii) inefficient coding.  Since I've got to fix the first, I may as well fix the second as well. In `lauxlib.c:luaL_findtable ()` the third argument is a table name which can be of the form `rotable.some_extremely_long_table`.  Lines 410:415 handle the embedded '.' case and in this case would pass a length of 7 rather than the strlen which is 34.  However at `lrotable.c:23` a strlen check is made against `LUA_MAX_ROTABLE_NAME` (32).  The rotable name length is 7 but the strlen is 34 so this test fails and the function `luaL_findtable()` will invalidly return NULL.  This strlen is also redundant in that the strnmp limits the comparison to the supplied length anyway.

The dumb coding in that this is a linear scan and hence will check n/2 false matches in general before a true match.  You want the if test in the loop to short circuit as which the least cost potent test, and so:

```patch
--- a/app/lua/lrotable.c
+++ b/app/lua/lrotable.c
@@ -20,10 +20,8 @@ extern const luaR_table lua_rotable[];
 void* luaR_findglobal(const char *name, unsigned len) {
   unsigned i;
 
-  if (c_strlen(name) > LUA_MAX_ROTABLE_NAME)
-    return NULL;
   for (i=0; lua_rotable[i].name; i ++)
-    if (*lua_rotable[i].name != '\0' && c_strlen(lua_rotable[i].name) == len && !c_strncmp(lua_rotabl
+    if (lua_rotable[i].name[0] == name[0] && lua_rotable[i].name[len] == '\0' && !c_strncmp(lua_rotab
       return (void*)(lua_rotable[i].pentries);
     }
   return NULL;
```

The second test might cause unaligned read exceptions so this might run faster if I swap the terminator and strncmp tests around.

I am not going to issue a PR since I am doing other rotable stuff for this tasking interface and will sweep this up here.

jmattsson, 2016-01-14:
I'm probably missing something here, but if the longest a rotable name can be is 32 (presumably due to fixed array size), why is it invalid to return NULL if the scanned-for name is 34 long?

TerryE, 2016-01-14:
In my example the rotable is `rotable` which is 7 chars long, but the function returns NULL rather than the rotable address.

jmattsson, 2016-01-14:
Aaaah, right. Embedded dot. Now I'm with you.

TerryE, 2018-06-22:
I've never made this change!!  Added to my TODO.

closed 2019-04-10 by TerryE