nodemcu/nodemcu-firmware #930: Implement limits on time spent in lua so SDK has a chance to run

pjsg, 2016-01-14:
There are any number of issues arising from the fact that lua programs can consume all the CPU and not return often enough to the SDK. This leads to heap starvation as packets arrive (and also violates the SDK guidelines). It also appears to be (one of) the issue in #915 

I propose the following approach -- which I could undertake if there is general agreement on the approach.

The basic idea is to use the existing lua hook mechanism to get control from the VM every (say) 1000 instructions. If the time has exceeded 10ms since being invoked, then call lua_yield() and post a message so that the SDK will call back in the future. Once lua_yield is called, the call stack appears to unwrap itself, and control will return to the SDK. 

At this point, the SDK does its stuff and then handles the message previously posted. At this point it calls back into the node code, and this calls lua_resume to resume the suspended execution.

There are a number of flies in the ointment. For example, if a callback (from the network, or timer, or uart) is triggered, then there is another entry into the node code. I *suspect* that we want to run the new callback first (i.e. keep the existing callback handling codepath). Of course, that callback might exceed the time limit, so it too can get suspended. Etc.

This means that there needs to be a queue of suspended lua states. This gets difficult to manage if there are multiple suspended activities on the same 'L' object. These have to be run in reverse order. Other 'L' values can be run in any order (probably round robin for fairness). This does imply that there could be an impact on *which* 'L' value is used when callbacks are triggered. 

Since this code is going to be very close to the code that is going to handle the callbacks for gpio interrupts etc, it probably makes sense to do them in a particular order (to prevent merge issues). [As a side note, the Espressif documentation doesn't seem to say which functions can be called from an interrupt. I'm assuming that the system_os_post function can be.

My goal is to be able to have an infinite loop not bring down the system.

If the application is very badly behaved and lots of callbacks etc take a long time, then the system may run out of memory holding all the partial state, but it should panic cleanly. There is an interesting question as to what should happen if a timer callback (say triggered every 100ms) takes 1 second to execute. If we adopt the approach above, then we would build up more and more stack consumed on the 'L' instance used for the callback until we ran out of memory. There is an alternative 'run-to-completion' model where the callbacks are queued until there is no other work to do. However, in this case, an infinite loop would block everything from happening.

Anyway, I'm interested in thoughts as to whether this would work.

BTW The idea of using the hooks on line count is stolen from the eLua code which uses it for handling interrupts.

jmattsson, 2016-01-14:
My immediate reaction is that we cannot and should not try to do actual preemption plus what amounts to multi-threading if we allow callbacks to "skip the queue". Lua is not designed for that, and if nothing else it would be an absolute nightmare to explain to people how it works. Run-to-completion is sane (on a conceptual level, never mind the reality) and easy. Multi-threading is assuredly not.

If queuing up callbacks is an option, that could be interesting. Mostly though, I think as a first step it could be valuable to simply print warnings when Lua code hogs the CPU for too long, and perhaps optionally have it throw a Lua error on serious excessive CPU hogging.

I don't want to discourage you from digging into this space, but I think we need to remember that the end product is meant to make it easy for non-hardcore developers to use. To me, that very much excludes any notion of multi-threading (and I don't consider co-routines to fall into that category).

I'm sure Terry will have a lot more thoughts and ideas in area - I'm not a Lua internals expert by any reach.

TerryE, 2016-01-14:
In principle _if_ we could do it then is might make sense, but as Johny says there are quite a few major architectural impediments, as Lua is not designed to be preemptive.  The eLua mechanism is rather different because it assumes that the Lua firmware is sitting over "bare metal" as would be the case on an Arduino board.  We are not.  Espressif keeps a lot of the ESP8266 internals hidden and only expose a limited view and API through their SDK.  The main challenge as I see it is that we can't guarantee that if we do yield back to the SDK  then the next cb to be run might not be this compute task, but instead another timer alarm, input, or network callback all of which have been written assuming that they can start a Lua execution on L0.

Let me have a mull about this and post my thoughts later, but my instinct is that we either use a separate Lua thread for such long running activities, or if they are C code then treat them as entirely async to Lua and provide a cb hook.

pjsg, 2016-01-14:
[Which thread should we discuss this in?]

Anyway, I'm going to follow the suggestion above and put some monitoring in to see how long we hold the CPU before returning to the SDK.

pjsg, 2016-01-14:
file.format() took me 67 seconds
My app startup took 5 seconds (mostly compiling the lua files into .lc)
My timer handler seems to take around 450ms (this does refresh an attached i2c oled display)

I think that the "run-to-completion" model probably will surprise people the least.....

If you want to play, the code is in https://github.com/pjsg/nodemcu-firmware/tree/maxtime-in-lua -- based on the current dev branch. Just call node.tasktime() to get the time (in us) and the location of the max duration since the last call to tasktime.

```
> file.format()
format done.
> =node.tasktime()
67195681	task_lua
> 
```

```
> =node.tasktime()
454263	alarm_timer_common
```

nickandrew, 2016-01-16:
Would using the RTOS SDK help?

TerryE, 2016-01-16:
Uaaaarrrgghh.  Arnm, don't bring up the RTOS SDK.  If and when we support the esp32, then we are going to have to do this migration, but the whole underlying OS model is very different.  The RTOS is multithreaded and preemptive.  This isn't going to be a trivial piece of work.