nodemcu/nodemcu-firmware #933: WiFi AP scan does not return all channels/rssi an SSID is available on.

jmattsson, 2016-01-14:
I discovered this while verifying the #888 PR. The SDK scan happily returns multiple entries for an SSID, with one entry for each channel the SSID was found on. However, because we use a Lua table instead of an array to return the scan results, we only keep the latter of those scan entries.

This is really a very minor issue, and I'm not sure it's worth worrying about it especially since it would involve breaking backwards compatibility.

dnc40085, 2016-01-14:
@jmattsson I originally submitted PR #409 to fix this issue but then somebody complained that it broke their program, so I submitted PR #460 to allow changing between new format and old format to preserve backwards compatibility
old format (DEFAULT): (SSID : Authmode, RSSI, BSSID, Channel)
`wifi.sta.getap({ssid="Known_SSID"}, function(T) print("\r") for k,v in pairs(T) do print(k.." : "..v) end end)`
select new format: (BSSID : SSID, RSSI, Authmode, Channel)
`wifi.sta.getap({ssid="Known_SSID"}, 1, function(T) print("\r") for k,v in pairs(T) do print(k.." : "..v) end end)`

jmattsson, 2016-01-14:
I'm going to have to retest on the office network tomorrow - I *thought* it was the same BSSID on both channels...

jmattsson, 2016-01-20:
Nope, you're right, they've got different BSSIDs, and yes, the new format correctly deals with it. Between #460 and #888 everything appears to covered. Nothing to be seen here, sorry about the noise :)

closed 2016-01-20 by jmattsson