nodemcu/nodemcu-firmware #937: Creating a nodeMCU Lua official FAQ

TerryE, 2016-01-14:
If we are going to move the FAQ into the distro, then we can hardly call it unofficial (see #449 for previous issue).  This is to discuss the necessary changes to prepare the FAQ for moving into the official document set.

marcelstoer, 2016-04-10:
"How to write a new module" (inspiration from http://desperate-programmers.com/2016/01/19/creating-a-new-nodemcu-user-module/) would be a page of its own rather than an FAQ question, right?

TerryE, 2016-04-11:
Yes I t hunk that we need to split the FAQ into two parts: one for Lua developers and one for C module developers.

closed 2017-06-19 by marcelstoer

TerryE, 2017-06-19:
@marcelstoer Marcel, thanks for this.  What I suggest is that we open a new issue to role forward any open points and to act as a discussion framework for the next round of enhancements.  What would be good is if we could get an active discussion going with some of the more experienced Lua developers to get their feedback on the issues that trip them up and address these within a FAQ framework, but let's have this discussion on the new issue.

marcelstoer, 2017-06-19:
Sure, I wouldn't mind, feel free to open a new issue. For the stuff _I_ am interested in I opened and will continue to open dedicated issues (#1861 & #1862 for now).

PR #1899 (2017-06-19): Version 2.0 of the Lua Develoer FAQ
Fixes #937.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md)

This is a major rewrite of the Lua Developer FAQ to bring it up to date with the current state of development.

This type of FAQ is always going to tend to be the views of one author and in the case me.  The commnets I would be really interested in are:

-  Any typo or grammar fixes.
-  Points where you think that I am factually incorrect or have [... cut]