nodemcu/nodemcu-firmware #941: Clean up the HTTP client module

marcelstoer, 2016-01-15:
See https://github.com/nodemcu/nodemcu-firmware/pull/887#issuecomment-171936376:
- write documentation
- move examples from README to examples dir

jmattsson, 2016-01-16:
Somewhere/somewhen I volunteered to write the API docs for it. I've got it on my todo list, either this weekend or early next.

jmattsson, 2016-01-16:
I've added module docs to `newdocs`

jmattsson, 2016-01-16:
Considering the module docs include examples, should they perhaps just be removed from the readme? (as opposed to moved elsewhere)

devsaurus, 2016-01-16:
Guess so, #771 also proposes to remove examples from the README.
My approach is to have code *snippets* in the module docs, but to store full-fledged scripts that can actually be executed in `lua/examples`.

TerryE, 2016-01-16:
I think that the current code snippets can be very misleading.  I would prefer every module to have at least one full-fledged script which demonstrates best practice in using the module, and that this is in `lua_examples` _not_ `lua/examples`.

marcelstoer, 2016-01-16:
Indeed, as per #771 and #772 full-fledged examples, not to be mixed up with the snippets in the docs, should be removed from README.

PR #974 (2016-01-26): Transfer HTTP client examples to lua_examples/
Fixes #941
commit 08921c98 (2016-01-26): Transfer HTTP client examples to lua_examples/

Fixes #941
closed 2016-01-29 by marcelstoer