nodemcu/nodemcu-firmware #948: SD Card interfacing

laxmimerit, 2016-01-17:
Hello,
        where can I get sd card interfacing with esp8266 example in lua language?

TerryE, 2016-01-17:
This is the wrong place to ask this sort of Q. See #719 for links to forums which are aimed at providing applications programming help.  However we don't current provide a Lu programmable interface to standard SD cards, just the on-module Flash memory.

closed 2016-01-17 by TerryE

cjdg, 2016-04-01:
is there any plan to include an sd card interface?

TerryE, 2016-04-01:
Nope. Sorry.

marcelstoer, 2016-04-02:
> is there any plan to include an sd card interface?

It'd be nice to have I agree. Guess we'd welcome PRs of a ported SD lib like https://github.com/esp8266/Arduino/tree/master/libraries/SD. The WeMos folks prefer Arduino over NodeMCU and have Arduino examples for all their cool shields (-> Micro SD http://www.wemos.cc/Products/micro_sd_shield.html).

marcelstoer, 2016-06-22:
Arnim is looking into this #1321.