nodemcu/nodemcu-firmware #954: Review of new UART functionality added in the 1.x SDKs

TerryE, 2016-01-19:
As part of my work on sorting the tasking / ISR issues, I have been reviewing the various drivers.  Most are forks of the Espessif SDK 0.9x drivers an some with the odd enhancement.  The UART driver is different in that both the SDK 1.5 driver has had major rework by Espressif contributor and the nodemcu version also has also had -- largely divergent -- enhancements by @devsaurus, @jmattsson, @btsimonh and @nodemcu.  I don't want to add another stakeholder into this game.  I just suggest that one of you review the SDK 1.5 driver enhancements to see if there is any merit in merging any of these into our version.

I will be doing a minimum change PR to standardise task handling.

jmattsson, 2016-01-20:
When I did the 1.4.0 upgrade I did briefly have the 1.4.0 UART driver in, until I realized it was now GPL. It did not seem appropriate to force the rest of the NodeMCU codebase onto the GPL terms, so I backed it out and squashed that import.

I'm not aware of any lacking feature or issues with our UART driver as things stands, so I'd be inclined to leave it alone for now.

SerdarMSC, 2016-01-20:
Hi, i was burn latest version but i cant use uart mode 115200-8E1 mode. I have even parity problem, please help me for solution.

i use "uart.setup(0, 115200, 8, 1, 1, 0)" setting.

Thank you.

TerryE, 2016-01-31:
OK.  I'll close this as I see no reason to extend this discussion.

closed 2016-01-31 by TerryE