nodemcu/nodemcu-firmware #956: nodemcu firmware issue

BenjaminWongCK, 2016-01-20:
I used nodemcu firmware version 0.9.5 for a project for ESP8266. However, lots of issues with it and there was very limited memory left. So I decided to have a custom build firmware to solve the issues.

After I try two builds and upgrade the firmware, but new issues happened:
1. the 1st one is: the MAC address returned by command  wifi.ap.getmac() become low case, say  5e:cf:7f:81:c8:cf. Before is: 5E:CF:7F:81:C8:CF

2. the variable size limit is found to be 1024, but before is 2xxx. do we have any way to have big size variable?

After that I tried few different versions larger than 0.9.5.  but the issues are same.

Many tks!!
/Ben

eyaleb, 2016-01-20:
Looking at `master` branch.

1. in `sdk/esp_iot_sdk_v1.4.0/include/user_interface.h` you can see:
```
#define MACSTR "%02x:%02x:%02x:%02x:%02x:%02x"
```
I do not know what it was before, or if it changed, but you can define your preferred `MACSTR` in `app/include/user_config.h`, for example:
```
#define MACSTR "%02X:%02X:%02X:%02X:%02X:%02X"
```
and build your firmware.

TerryE, 2016-01-20:
The actual code block is
```C
#ifndef MAC2STR
#define MAC2STR(a) (a)[0], (a)[1], (a)[2], (a)[3], (a)[4], (a)[5]
#define MACSTR "%02x:%02x:%02x:%02x:%02x:%02x"
#endif
```
So you need to define both `MAC2STR` and `MACSTR` in your `user_config.h`.  But this is a config issue and not a bug.

closed 2016-01-20 by TerryE

BenjaminWongCK, 2016-01-21:
Understood that is not a bug. Wonder why had such change and in fact, that
would cause people time to address it.
I spent  few hours yesterday to address it by changing few firmware, the
finding is that :
from the version dated 20150625, there were some changes might be applied.
So I fallback to version 20150406 for the time being..

Many tks!!!

jmattsson, 2016-01-21:
This was a change made by Espressif in the SDK, not by the NodeMCU project.