nodemcu/nodemcu-firmware #957: SOFTAP mode unable to connect

BenjaminWongCK, 2016-01-20:
II tried to ungraded nodemcu firmwares from  "nodemcu_float_0.9.6-dev_20150331.bin"
to the latest version posted on the web on yesterday and today. It is found that:
from "nodemcu_float_0.9.6-dev_20150625.bin" and onward to the latest version seemed not stable when launch SoftAp mode. 

The symptom is: the AP could be found by PC/smart phone, but unstable either being disconnected in few second or unable to connect.

once I fallback to  "nodemcu_float_0.9.6-dev_20150406.bin" or  backward version, everything is back to normal.

Many tks!!

ABonner, 2016-01-21:
Please try the NodeMCU Build tool at http://nodemcu-build.com/ instead of using the quite outdated 0.9.6. It may solve your issue.

closed 2016-01-22 by TerryE

BenjaminWongCK, 2016-01-22:
Adam,

Thanks for your advice. I have tried NodeMCU Build tool and got two built
versions and test them. The unstable issues of softAP have been gone,
however, after taking a whole morning testing, I got another issue when
sending data to web browsers. The issue is: only the *1st line* of data
could be sent; the rest could NOT be sent.  Here below is the code.

-------------------------------------------------------------------------------
-- Start a simple http server
srv=net.createServer(net.TCP)
srv:listen(80,function(conn)
   conn:on("receive",function(conn,payload)
     print(payload)

 -- start sending three line of data
    conn:send('<h1>ESP8266 project testing..</h1>\n')
    conn:send('<h2> test line 2 - Hello, NodeMCU!!! </h2>\n')
    conn:send('<h3> test line 3 - Hello, NodeMCU!!! </h3>')

  end)
  conn:on("sent",function(conn) conn:close() end)
end)
--------------------------------------------------------------------------------

The browser only display the 1st line  "*<h1>ESP8266 project testing..</h1>*
"

Now, I fallback to the version dated "20150406" and it work fine but with
limited memory left and sometimes hang-up.

many tks!!

ABonner, 2016-01-22:
@BenjaminWongCK 

You can't do multiple sends and have them all go through. You need to use callbacks on the first send to trigger the next ones.

See for more information:
http://nodemcu.readthedocs.org/en/dev/en/modules/net/#netsocketsend

BenjaminWongCK, 2016-01-22:
Oh...

Multiple send worked before and I now use version 20150406 and it works
fine.

Anyway, I will appreciate it if you can give me an example coding for
sending multiple data since I could not find any from the website you
provided.

many tks!
/Ben

ABonner, 2016-01-22:
@BenjaminWongCK Read the related conversation in this issue and follow the links. Espressif changed the behavior of socket:send in more recent SDK.

https://github.com/nodemcu/nodemcu-firmware/issues/964

BenjaminWongCK, 2016-01-22:
Thanks Adam,

I just went through it, but it seems complicated, I may take time to get
familiar with it and change codes..

However, I got another issue with the newer firmware on Telnet . Before I
could Telnet to send command to ESP but with the newer firmware, only the
1st line sent to my Telnet client if multiple line response from ESP( ie.
issue  "list file" command). I think the problem also with "multiple send"

The Telnet code was same as the sample code on the web of nodemcu (
https://nodemcu.readthedocs.org/en/dev/en/faq/#how-is-coding-for-the-esp8266-different-to-standard-lua
)

The change of prohibition of Multiple Send make lots of issue to me and
others as learnt from the support post, could you please consider or
escalate to the personnel concerned see if the function of "Multiple Send"
function could be resumed.

Normally, a single response to a HTTP request involve multiple lines of
header followed by a HTML file. How come the change of new firmware doesn't
consider such HTTP behavior.....

Many tks!!