nodemcu/nodemcu-firmware #959: POST with net.socket don't send body

apmascalabs, 2016-01-20:
Hi, 
I'm tryng to send a POST with nodeMCU custom firmware build on the cloud with net module.
The server is neo4j.
This is a part of the script:

 post_s="POST /db/data/node HTTP/1.1\r\n"
    .."Host: projects.mascalabs.com:7474\r\n"
    .."Connection: keep-alive\r\n"
    .."Accept:application/json, text/plain, */*\r\n"
    .."X-stream: true\r\n"
    .."Authorization : Basic xxxxxxxxxxxxxxxxxxxx\r\n"
    .. "Content-Type: application/json\r\n\r\n"
    .."{'this':'one'}"
    
    conn = nil
    conn = net.createConnection(net.TCP,0)
    conn:on("receive",function(conn,payload)success=true print(payload)end)
    conn:connect(7474,'projects.mascalabs.com')
    conn:on("connection",function(conn,payload)print("Connected")
     conn:send(post_s)
   
The node is created because I'm able to connect with the correct auth, but there is no body in my POST.
the answer from server is the creation of a node but without any 'data'.
I've tryied many kind of query following the neo4j manual and the cypher manual,I've tryied also send to 'httpbin' but the body is always empty.
Which is the correct syntax to send POST complete of body?
Is POST really supported from nodeMCU?
thanks

pjsg, 2016-01-20:
There are two things that jump out at me:

* The body is not actually JSON -- JSON uses " rather than ' to delimit strings
* There is no content-length header.

I don't if these are your problems... What happens if you take the same datablock and use something like netcat to send it to your server?

jmattsson, 2016-01-20:
I believe @pjsg already covered the main point (bad json). May I also suggest that you consider using [http.post()](http://nodemcu.readthedocs.org/en/dev/en/modules/http/#httppost) to simplify matters a bit for you? The `http` module is available in the `dev` branch.

TerryE, 2016-01-20:
Sorry but this issue falls outside the scope of this list.  Please read #719 for suggestions on where to post this sort of question.

closed 2016-01-20 by TerryE

apmascalabs, 2016-01-21:
Ok , thank s to pjsg and jmattson for help.
I'm really sorry that this project stop.

PS: 
I've add the content length and substitute ' with \" but nothing change.
Thanks

vkolarov, 2016-01-30:
I can reproduce this using Safari (on Desktop or iPhone). Just post a simple html form (with an input field) with Safari. When you print the payload on the ESP put a leading and trailing character before and after the payload. F.g. the pipe character:

print("payload: |" .. payload .. "|")

Here is the output:

payload: |POST /savesettings HTTP/1.1
Host: 192.168.4.1
Content-Type: application/x-www-form-urlencoded
Origin: http://192.168.4.1
Content-Length: 15
Connection: keep-alive
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_11_3) AppleWebKit/601.4.4 (KHTML, like Gecko) Version/9.0.3 Safari/601.4.4
Referer: http://192.168.4.1/savesettings
Accept-Language: en-us
Accept-Encoding: gzip, deflate

|
payload: |ssid=&password=|

There are two different connections: one containing the header and one containing the body.
ATTENTION: If you omit the leading and trailing characters in the print statement, the header and body are printed one after another and this can make you thing there is only one payload.

Some browsers (or clients like neo4j) are sending post data in two or more different TCP IP packets. One with the header and another one (or more) with the body data. You can read more on this [here](https://josephscott.org/archives/2009/08/xmlhttprequest-xhr-uses-multiple-packets-for-http-post/).

It seems that the ESP closes the connection after the first packet. And then opens another connection for the second packet.
Chrome f.g. sends the header and body together in one packet and POST works as expected.

nickandrew, 2016-01-31:
It's unclear if your send call is part of the on("connection") function callback.

vkolarov, 2016-01-31:
I'm using on("receive") and not on("connection") event.

```
wifi.setmode(wifi.SOFTAP)
cfg = {}
cfg.ssid = "ESP_8266"
cfg.auth = wifi.AUTH_OPEN
wifi.ap.config(cfg)
print(wifi.ap.getip())

srv=net.createServer(net.TCP,30)
srv:listen(80, function(conn)
    conn:on("receive", function(conn, payload)
        print("payload: |" .. payload .. "|")
        conn:send("HTTP/1.1 200 OK\r\nServer: NodeLuau\r\nContent-Type: text/html\r\nConnection: close\r\n\r\n")
        conn:send("<!DOCTYPE HTML>")
        conn:send("<html><head><title>test</title></head>")
        conn:send("<body><form method=\"post\" action=\"/savesettings\">")
        conn:send("WiFi SSID: <input type=\"text\" name=\"ssid\" /><br>")
        conn:send("WiFi Password: <input type=\"password\" name=\"password\" /><br>")
        conn:send("<input type=\"submit\" value=\"Save\" />")
        conn:send("</form></body></html>")
    end)
    conn:on("sent", function(conn)
        conn:close()
        collectgarbage()
    end)
end)
```

Kind regards,
Veselin

nickandrew, 2016-01-31:
You can't do that with current firmware. After the first conn:send() call, subsequent calls to conn:send() need to be within a conn:on("sent") function hook.

apmascalabs, 2016-01-31:
I solved simply change the header 'Connection: keep-alive ' with 'Connection: close' and now I'm able to send all kind of query with the correct syntax requested from the server.
Thanks

apmascalabs, 2016-01-31:
But I must use \" in the query.

vkolarov, 2016-01-31:
Hello nickandrew,

where did you read that each subsequent conn:send() should be in a :on("sent") function?
And why do you think this is not possible with the current firmware? Did you try my example above?
I've tested it with Safari, Chrome and Firefox and it is working.
And how is this related with the form POST problem?

vkolarov, 2016-01-31:
Hi apmascalabs,

can you test to POST some form data with Safari?

apmascalabs, 2016-01-31:
Hi vkolarov , I don't understand your question. I had have problem to send POST to my server with nodeMCU used like a client in Station mode. The problem was the headers requested from server. I'm working on a project with a server created from my collaborator, we use esp8266 like client, so we need to send query post data to this server . 
We are deciding if use nodemcu fw anymore after the Terry's answer or change fw .

nickandrew, 2016-01-31:
@vkolarov Read the documentation on net.socket:send() ...

> Multiple consecutive send() calls aren't guaranteed to work (and often don't) as network requests are
> treated as separate tasks by the SDK. Instead, subscribe to the "sent" event on the socket and send
> additional data (or close) in that callback. See #730 for an example and explanation.

vkolarov, 2016-01-31:
Sorry nickandrew! I misunderstood, my bad. My problem is on the other way. I'm using the ESP8266 as web server. And I'm POSTing the data from Safari to the ESP8266. The ESP8266 receives two payloads one with the header data only and one with the body data only.

kilhankulhe, 2016-06-29:
How can I Post something with JSON format, and with X-Auth-Token as Header?

marcelstoer, 2016-06-30:
@vkolarov pls report it as a separate issue, someone else reported the same on esp8266.com: http://www.esp8266.com/viewtopic.php?f=18&t=9740